carla-simulator/carla #6009: How to change WalkerBoneControl via. spawn_npc.py?

laurenceandrews, 2022-12-04:
I'm trying to spawn a group of walkers with edited skeletons without using UnrealEditor. I'd love to be able to use the spawn_npc.py example file in the PythonAPI, but since it uses WalkerAIController class to spawn NPCs, I'm struggling to figure out how to use the Walker class's apply_control method to apply my bone transforms.

Clearly there's a type mismatch here, but I was hoping that somehow I might be able to fudge this to work, perhaps using the Actor class' parent method. The script runs when I create a WalkerBoneControl and apply it to the parent of my WalkerAIControllers, however I'm not seeing any difference in the walkers when I run the script. Any advice?

At the moment I'm trying to put my code in step 5, where the controllers are initialized, as follows:

# 5. initialize each controller and set target to walk to (list is [controller, actor, controller, actor ...])
        # set how many pedestrians can cross the road
        world.set_pedestrians_cross_factor(percentagePedestriansCrossing)
        for i in range(0, len(all_id), 2):
            # start walker
            all_actors[i].start()
            # set walk to random point
            all_actors[i].go_to_location(world.get_random_location_from_navigation())
            # max speed
            all_actors[i].set_max_speed(float(walker_speed[int(i/2)]))
        
            # attempt to apply bone control to Walker via. WalkerAIController (my code below, expecting straight arms)
            control = carla.WalkerBoneControl()
            first_tuple = ('crl_arm__L', carla.Transform(rotation=carla.Rotation(roll=180)))
            second_tuple = ('crl_arm__R', carla.Transform(rotation=carla.Rotation(roll=180)))
            third_tuple = ('ctrl_shoulder__L', carla.Transform(rotation=carla.Rotation(roll=180)))
            fourth_tuple = ('ctrl_shoulder__R', carla.Transform(rotation=carla.Rotation(roll=180)))
            control.bone_transforms = [first_tuple, second_tuple, third_tuple, fourth_tuple]
            all_actors[i].parent.apply_control(control)

HMX2013, 2025-05-13:
> I'm trying to spawn a group of walkers with edited skeletons without using UnrealEditor. I'd love to be able to use the spawn_npc.py example file in the PythonAPI, but since the script uses the WalkerAIController class to spawn NPCs, I'm struggling to figure out how to use the Walker class' apply_control method to apply my bone transforms.
> 
> Clearly there's a type mismatch here, but I was hoping that somehow, I might be able to fudge this to work, perhaps using the Actor class' parent method, since Walker and WalkerAIController are inherited from Actor. The script runs when I create a WalkerBoneControl and apply it to the parent of my WalkerAIControllers, however I'm not seeing any difference in the walkers when I run the script. Any advice?

Have you solved this problem?  I'm struggling with it too.

Blyron, 2026-11-12:
Bones are applied to the walker actor, not to its `controller.ai.walker`. Use `walker.set_bones(carla.WalkerBoneControlIn(...))` and then `walker.blend_pose(1.0)`; see [Walker skeletons](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/tuto_walker_skeletons.md). Closing.

closed 2026-11-12 by Blyron