carla-simulator/carla #6020: some problem during make pythonAPI

TeleYuhao, 2022-12-08:
some problem had happened during building
the warning are as follow:
ninja: error: '../boost-1.72.0-c8-install/lib/libboost_filesystem.a', needed by 'LibCarla/cmake/test/libcarla_test_client_release', missing and no known rule to make it
make: *** [Util/BuildTools/Linux.mk:124：LibCarla.client.release] 

[image]

emanef13, 2023-06-10:
Hey, did you manage to solve it?
I have the same issue

TeleYuhao, 2023-06-10:
I try it again and again .At last I put the file folder of some part of carla in the cmake file and turn the vpn on , make it again, finally solve the problem

emanef13, 2023-06-10:
I just managed to solve it too. 
I rebooted my computer.
I removed any exported python path from .bashrc file.
I reinstalled CARLA from source (as my initial installation was)
And I was able to execute make PythonAPI with no problems :)