carla-simulator/carla #6038: carla_msgs problem

6Lackiu, 2022-12-12:
Hello! 
When I use carla ros2, run the command `ros2 launch carla_ ros_ bridge carla_ ros_ Bridge.launch.py` encountered the following error:

`[INFO] [launch]: All log files can be found below /home/liu/.ros/log/2022-12-12-20-43-45-074309-ub20liu-65603
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [bridge-1]: process started with pid [65605]
[bridge-1] [INFO] [1670849026.578995010] [carla_ros_bridge]: Trying to connect to localhost:2000
[bridge-1] [INFO] [1670849026.782688607] [carla_ros_bridge]: Loading town 'Town01' (previous: 'Carla/Maps/Town10HD_Opt').
[bridge-1] [INFO] [1670849028.779009110] [carla_ros_bridge]: synchronous_mode: True
[bridge-1] [INFO] [1670849028.779256029] [carla_ros_bridge]: fixed_delta_seconds: 0.05
[bridge-1] [INFO] [1670849028.812487498] [carla_ros_bridge]: Parameters:
[bridge-1] [INFO] [1670849028.812704073] [carla_ros_bridge]:   host: localhost
[bridge-1] [INFO] [1670849028.812891873] [carla_ros_bridge]:   port: 2000
[bridge-1] [INFO] [1670849028.813041475] [carla_ros_bridge]:   timeout: 2
[bridge-1] [INFO] [1670849028.813192936] [carla_ros_bridge]:   passive: False
[bridge-1] [INFO] [1670849028.813339817] [carla_ros_bridge]:   synchronous_mode: True
[bridge-1] [INFO] [1670849028.813487533] [carla_ros_bridge]:   synchronous_mode_wait_for_vehicle_control_command: False
[bridge-1] [INFO] [1670849028.813639043] [carla_ros_bridge]:   fixed_delta_seconds: 0.05
[bridge-1] [INFO] [1670849028.813784656] [carla_ros_bridge]:   register_all_sensors: True
[bridge-1] [INFO] [1670849028.813930238] [carla_ros_bridge]:   town: Town01
[bridge-1] [INFO] [1670849028.814085970] [carla_ros_bridge]:   ego_vehicle: {'role_name': 'heroego_vehiclehero0hero1hero2hero3hero4hero5hero6hero7hero8hero9'}
[bridge-1] [INFO] [1670849028.850291119] [carla_ros_bridge]: Shutting down...
[bridge-1] Traceback (most recent call last):
[bridge-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
[bridge-1]     return importlib.import_module(module_name, package=pkg_name)
[bridge-1]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[bridge-1]     return _bootstrap._gcd_import(name[level:], package, level)
[bridge-1]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[bridge-1]   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
[bridge-1]   File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
[bridge-1] ModuleNotFoundError: No module named 'carla_msgs.carla_msgs_s__rosidl_typesupport_c'
`

Main problem is: `ModuleNotFoundError: No module named 'carla_msgs.carla_msgs_s__rosidl_typesupport_c'`

How to solve this problem? Thank you!

Carla: 0.9.13
ROS2 foxy
Ubuntu20.04

fatemehprhm, 2023-01-09:
Hi
I had the same problem, in my case it was caused by confliction in conda environment, I did the following steps to fix it.

1. First of all deactivate all conda environments, even ''base'' by `conda deactivate`
2. Then open bashrc or zshrc depending on which shell you use: `sudo gedit ~/.bashrc`
3. Type `export $PATH` in terminal to get the search directories. it will probably be something like this depending on what ros version you are using and your computer name:
`"/opt/ros/foxy/bin:/home/vandv/anaconda3/condabin:/home/vandv/anaconda3/envs/carla/bin:/home/vandv/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"`
4. Copy and paste it in bashrc file in front of `export PATH="<copy here>"` and remove this part `"/home/vandv/anaconda3/envs/carla/bin"` of it like below: 
`export PATH="/opt/ros/foxy/bin:/home/vandv/anaconda3/condabin:/home/vandv/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"`
5. Then save and close bashrc file and reboot or type: `source ~/.bashrc`
6. Then remove build file of carla ros bridge in carla-ros-bridge folder and in another terminal build your carla ros bridge in your workspace again by `colcon build`.
7. Open another terminal and do the steps of "Run the ROS bridge" part again.

I hope it'll work for you.

6Lackiu, 2023-01-09:
> Hi I had the same problem, in my case it was caused by confliction in conda environment, I did the following steps to fix it.
> 
> 1. First of all deactivate all conda environments, even ''base'' by `conda deactivate`
> 2. Then open bashrc or zshrc depending on which shell you use: `sudo gedit ~/.bashrc`
> 3. Type `export $PATH` in terminal to get the search directories. it will probably be something like this depending on what ros version you are using and your computer name:
>    `"/opt/ros/foxy/bin:/home/vandv/anaconda3/condabin:/home/vandv/anaconda3/envs/carla/bin:/home/vandv/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"`
> 4. Copy and paste it in bashrc file in front of `export PATH="<copy here>"` and remove this part `"/home/vandv/anaconda3/envs/carla/bin"` of it like below:
>    `export PATH="/opt/ros/foxy/bin:/home/vandv/anaconda3/condabin:/home/vandv/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"`
> 5. Then save and close bashrc file and reboot or type: `source ~/.bashrc`
> 6. Then remove build file of carla ros bridge in carla-ros-bridge folder and in another terminal build your carla ros bridge in your workspace again by `colcon build`.
> 7. Open another terminal, activate whatever conda environment you were using and do the steps of "Run the ROS bridge" part again.
> 
> I hope it'll work for you.

Thanks! I've solved the problem, it's indeed conda problem.
Still, thank you very much for your generous answer!

SovereignKing, 2023-11-13:
> Thanks! I've solved the problem, it's indeed conda problem. Still, thank you very much for your generous answer!

excuse me,I met the same problem and I have installed miniconda,could you please tell me how to solve it?Thank you very much!

fatemehprhm, 2023-11-13:
@SovereignKing Hi, sorry but I don't have any experience with miniconda. I don't know if there is any fundumental differences between conda and miniconda so maybe you can try the approach I mentioned with your path specifications and see if it works.

xiboli, 2024-01-10:
Although it is already a closed issue, still I would provide my solution.
I have observed that this happens quite often when we compile carla_msgs in anaconda enviroment. So the solution is, deactivate all the anaconda environment and delete all related messages in build and install. Then rebuild it again, this problem will not happen.