carla-simulator/carla #6043: Segfault on dtNavMesh::getPolyHeight

truncs, 2022-12-13:
CARLA version: dev
Platform/OS: Linux
Problem you have experienced:
On consecutive map loads for and then adding a few walker controllers leads to a segfault. I have observed that this specifically happens when loading map Town10HD and then loading Town02. The trace is below

```
#0  0x00007ffeda750e33 in dtNavMesh::getPolyHeight(dtMeshTile const*, dtPoly const*, float const*, float*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#1  0x00007ffeda75145e in dtNavMesh::closestPointOnPoly(unsigned int, float const*, float*, bool*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#2  0x00007ffeda7562c6 in dtNavMeshQuery::closestPointOnPoly(unsigned int, float const*, float*, bool*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#3  0x00007ffeda76065e in dtFindNearestPolyQuery::process(dtMeshTile const*, dtPoly**, unsigned int*, int) () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#4  0x00007ffeda75729d in dtNavMeshQuery::queryPolygonsInTile(dtMeshTile const*, float const*, float const*, dtQueryFilter const*, dtPolyQuery*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#5  0x00007ffeda756a25 in dtNavMeshQuery::queryPolygons(float const*, float const*, dtQueryFilter const*, dtPolyQuery*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#6  0x00007ffeda75670c in dtNavMeshQuery::findNearestPoly(float const*, float const*, dtQueryFilter const*, unsigned int*, float*) const () from /home/aditya/.local/lib/python3.6/site-packages/carla/libcarla.cpython-36m-x86_64-linux-gnu.so
#7  0x00007ffeda5b38b8 in carla::nav::Navigation::GetAgentRoute (this=0x221fc50, id=<optimized out>, from=..., to=..., path=std::vector of length 0, capacity 0, area=std::vector of length 0, capacity 0) at ../../LibCarla/source/carla/nav/Navigation.cpp:409
#8  0x00007ffeda5b7e3b in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=822, to=...) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:133
#9  0x00007ffeda5b7c8f in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=<optimized out>) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:107
#10 carla::nav::WalkerManager::SetWalkerNextPoint (this=0x221fdb0, id=822) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:195
#11 0x00007ffeda5b7ffa in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=822, to=...) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:162
#12 0x00007ffeda5b7c8f in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=<optimized out>) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:107
#13 carla::nav::WalkerManager::SetWalkerNextPoint (this=0x221fdb0, id=822) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:195
#14 0x00007ffeda5b7ffa in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=822, to=...) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:162
#15 0x00007ffeda5b7c8f in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=<optimized out>) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:107
#16 carla::nav::WalkerManager::SetWalkerNextPoint (this=0x221fdb0, id=822) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:195
#17 0x00007ffeda5b7ffa in carla::nav::WalkerManager::SetWalkerRoute (this=<optimized out>, id=822, to=...) at ../../LibCarla/source/carla/nav/WalkerManager.cpp:162
``` 
Steps to reproduce:

Load Town10 and start a simulation with 100 walkers. Then load Town02 and then start a simulation with 100 walkers. Do this multiple times and eventually you will end up getting a  segfault.

Other information (documentation you consulted, workarounds you tried):
I reproduced this on another machine so hardware doesn't seem to be the issue.

DanielCoelho112, 2022-12-24:
Hi  @truncs,

Hi encountered the same error a few weeks ago.  
In my case, the problem was that the z coordinate of the walker position was negative, and then when I did controller.go_to_location(.) that error was thrown.

My workaround was before calling controller.go_to_location(.), I check the z coordinate of the walker, and if it is negative I don't call the method controller.go_to_location. 

It is not a perfect solution, but at least it avoids the error.

adwiii, 2023-06-30:
I had posted [this comment](https://github.com/carla-simulator/carla/issues/4350#issuecomment-1613272077) on #4350, but I think it may be related to this issue instead. I have not tried compiling from source to find where the segfault occurs in the C++ code. However, the Z coordinate trick does not fix the issue - I am finding segfaults even when the Z index is not negative. In one segfaulting run, these are the values that lead to segfault:

```
walker location: Location(x=400.612518, y=206.038651, z=1.974510)
walker controller location: Location(x=400.612518, y=206.038651, z=1.974510)
going to location: Location(x=6.309162, y=220.257812, z=0.000000)
```

faizansana, 2023-08-24:
> I had posted [this comment](https://github.com/carla-simulator/carla/issues/4350#issuecomment-1613272077) on #4350, but I think it may be related to this issue instead. I have not tried compiling from source to find where the segfault occurs in the C++ code. However, the Z coordinate trick does not fix the issue - I am finding segfaults even when the Z index is not negative. In one segfaulting run, these are the values that lead to segfault:
> 
> ```
> walker location: Location(x=400.612518, y=206.038651, z=1.974510)
> walker controller location: Location(x=400.612518, y=206.038651, z=1.974510)
> going to location: Location(x=6.309162, y=220.257812, z=0.000000)
> ```
> 
> Edit: when I am seeing this issue does not appear to have any basis with spawning walkers in different maps. Even if I start with a certain town loaded and reload _that same town_, it still hits the segfault.

Were you able to find a way around this?

adwiii, 2023-08-24:
> Were you able to find a way around this?

I have not found a way to solve this other than trying to avoid the segfault. I did not continue investigating.

faizansana, 2023-08-24:
> > Were you able to find a way around this?
> 
> I have not found a way to solve this other than trying to avoid the segfault. I did not continue investigating.

Do you mind expanding on what you mean by "avoiding the segfault". Do you not use the walker controller anymore?

adwiii, 2023-08-24:
Yes, I have not used the walker controller in any meaningful way since I posted here, at least not in general. I was able to spawn a handful at a time when testing things, but it was not robust enough for me to use long-term so I stopped exploring it and haven't gotten back around to it yet.

I believe other projects may get around this issue by using [ScenarioRunner](https://carla-scenariorunner.readthedocs.io/en/latest/) which has its own separate [pedestrian control](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/actorcontrols/pedestrian_control.py). I have set up demos of projects that I believe are using this code and have pedestrians walking short distances, though I have not inspected the code to see how they are accomplishing that.

faizansana, 2023-08-24:
Oh okay, that makes sense. I am using it for RL and the environment continuously seg faults which is getting pretty annoying so I have been investigating but looks like I might have to use ScenarioRunner.

Thanks for the suggestion!