carla-simulator/carla #6050: Adding custom vehicles to carla package

chrisvdploeg, 2022-12-16:
Hi,

I'm looking for a way to import custom vehicles into the carla (binary) package. Reason for this is that building from source and running carla through the unreal editor generates a lot of unnecessary and unneeded overhead. So far, I see that ingestion in the package is an option for custom maps and static (?) props, however, it would be great if we could do the same with vehicles.

I have managed to import a custom vehicle in the Carla build from source, which results in a couple of uasset files as well as an adjustment of the "VehicleFactory". Is there some way to ingest these files into the Carla package? Or, alternatively, create a new package which I can distribute to colleagues for them to use a package version of Carla including the custom vehicles?

Thank you.

Gowtham-Raj-M, 2023-09-04:
@chrisvdploeg , Do you find any solution to add dynamic objects to carla package?

banubuyuker, 2023-12-07:
@Gowtham-Raj-M, have you found a solution?

matejm42, 2024-01-03:
I was able to generate a standalone package according to [docs](https://carla.readthedocs.io/en/latest/tuto_A_create_standalone/) by running
```
make package ARGS="--packages=Carla"
```
from the Carla folder. But this builds the whole Carla binary, I don't know if it's possible to generate package with only added assets and import them into already used build.

Blyron, 2026-11-12:
This is now possible with content packs: you can add your own vehicles (and maps, props and walkers) to a packaged CARLA without rebuilding it. See [Content packs](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_packs.md) and [Custom vehicles](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/content_authoring_vehicles.md). Closing as implemented.

closed 2026-11-12 by Blyron