carla-simulator/carla #6062: How to add the maps  of Town06 and 07 to the carla?

wx983, 2022-12-24:
Version: 0.9.12
OS: Ubuntu 18.04
Built from source

I have downloaded the Additional_maps_0.9.12. But now I don't know how to import the maps.

vinaykumarhegde, 2023-01-04:
Copy them over to the `Import` directory and run `ImportAssets.sh` script [doc](https://carla.readthedocs.io/en/latest/start_quickstart/#import-additional-assets)

Madecu, 2023-01-16:
It is exactly how vinaykumarhegde mentioned.

Thanks vinaykumarhegde.

I'll close the issue

closed 2023-01-16 by Madecu

NayaBaslan, 2024-10-17:
can you provide more information about how to then load the additional maps.
Usually for the ones already available, i use
`python3 config.py --map Town01`

But the option for additional maps doesnt seem available.
Can you provide me with the right command I need to run to change the map to one of the additional maps?
I am using CARLA 0.9.15 and need to use Town12