carla-simulator/carla #6068: Simulator crashed (Signal 11 Failed to find symbol file "CarlaUE4-Linux-Shipping.sym")

MoritzNekolla, 2022-12-27:
CARLA version: 0.9.13
Platform/OS: Linux

Error-Message: 
```
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Steps to produce:
I run carla server side by:  
```
./CarlaUE4.sh -RenderOffScreen -opengl --carla-world-port=2000
```
There are two ways to produce the error:

1.
then I create a world(client) and connect it to the server. Afterwards, I take an image from the scene (rgb camera on car) and delete the world (client). I then create another world and repeat the process several times. Code example:
```python
for x in range(10000):
    env = Environment(host="localhost", port=2000)
    env.init_ego()
    obs_current = env.reset()
    saveObservation(obs_current)
```
Around the 20th world the error (mentioned above) occurs.

2.
If I only generate one single client and then call env.reset() around 2k times, the same error occurs.
```python
env = Environment(host="localhost", port=2000)
env.init_ego()
for x in range(10000):
    obs_current = env.reset()
    saveObservation(obs_current)
```

Why does this happen? I checked out [this](https://github.com/jbnunn/CARLADesktop/issues/6#issue-744186460) but in their case they cannot even start the carla server, which works fine in my case.
I did monitor the GPU/CPU capacity and I can confirm that there is no extensive memory usage due to wrongly destroyed actors or clients.

AliElKhalil, 2023-10-18:
Hi, I have the same issue when running Carla with 

`./CarlaUE4.sh -windowed -ResX=320 -ResY=240 -benchmark -fps=10 -quality-level=low>
`

It crashes when I am running a scenic scenario in it, the ego vehicule moves forward in its lane for a few, then I have the same error message.

Did you manage to resolve the issue ?

zjfhi2008, 2023-12-27:
I had the same problem, does any people has solution?

mrsgzg, 2024-03-21:
I met too, when I wanted to destroy the traffic light

lichengguang, 2025-01-14:
I met too.when i want load a xodr