carla-simulator/carla #6072: Sumo Co-simulation is not running (Error: tcpip::Socket::recvAndCheck @ recv: peer shutdown)

vsale, 2022-12-30:
Dear all,

our Sumo Co-Simulation is not running. We dont know why?
We have following Error Report:

Error: tcpip::Socket::recvAndCheck @ recv: peer shutdown

We are using Carla UE4 0.9.13 and Sumo 1.15.0 on Windows machine.

Can someone help?

Best

jy723, 2022-12-30:
1.change the file
/Co-Simulation/Sumo/sumo_integration/sumo_simulation.py
304    # sumo_net = traci.sumolib.net.readNet(net_file)
         sumo_net = sumolib.net.readNet(net_file)
because the sumo tools/traci/__init_.py change
2.you can try

hlwei, 2023-04-19:
> 

Fix my problem, when carrying the co-simulation of Carla and Sumo. Thx.

Blyron, 2026-11-12:
SUMO and PTV-Vissim co-simulation were part of the Unreal Engine 4.26 line and are not in the Unreal Engine 5 builds (see [Traffic simulation](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_traffic.md)). Closing, as the UE4 line is no longer maintained.

closed 2026-11-12 by Blyron