carla-simulator/carla #6074: N-wheeled vehicle from release blog post

Sid1057, 2023-01-01:
Hello! First of all thanks for the last release!

I am very interested in the car from carla release blog post.

I thought it would be in release, but after checking the blueprint info and outputting the vehicle gallery script, I realize that this vehicle is not integrated into the current release.

Will you add them in some patch? Or add an example to the documentation with resources for it, how to add it?

[image]

KPicherit, 2023-01-16:
I'm also interested in this model !

Madecu, 2023-01-26:
Hi @Sid1057 ,

The 6 wheeled truck has been added into Carla source. You can download it from there.

Thanks for open an issue here and I apologize for the answer delay.

Sid1057, 2023-01-26:
@Madecu thanks a lot!

closed 2023-01-26 by Sid1057

KPicherit, 2023-01-31:
Got it, thanks a lot ! :+1: