carla-simulator/carla #6075: Unreal Engine Version 4.26.2 Crashes Upon Project Creation

doom6461, 2023-01-02:
Hello, I am following the guide in the link below to build carla 0.9.13 for windows:
https://carla.readthedocs.io/en/0.9.13/build_windows/

I have a Windows 10 PC with a NVIDIA Geforce RTX 2060 graphics card.
I followed all the steps listed in the document above and cloned Unreal Engine from the following address (cloning from this address gives me UE_4.26.2):
https://github.com/CarlaUnreal/UnrealEngine.git

I ran the Setup.bat and GenerateProjectFiles.bat scripts successfully.
I opened up UE4.sln with Microsoft Visual Studio 2019 and selected the ‘Development Editor’, ‘Win64’ and ‘UnrealBuildTool’ options.
I built the UE4 solution successfully.

Now that everything is complete, when I run UE4Editor.exe I can create a new project. But when the project opens, the editor suddenly crashes with the following message:

> Fatal error: [File:C:/Users/123/Downloads/ue4/Engine/Source/Runtime/RenderCore/Private/RenderingThread.cpp] [Line: 907]
Rendering thread exception:
Assertion failed: ResourceTableFrameCounter == INDEX_NONE [File:C:/Users/123/Downloads/ue4/Engine/Source/Runtime/Windows/D3D11RHI/Private/D3D11Texture.cpp] [Line: 2260]

> 0x00007ff8c1bccd29 KERNELBASE.dll!UnknownFunction []
0x00007ff85e3cac76 UE4Editor-Core.dll!ReportAssert() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Windows\WindowsPlatformCrashContext.cpp:1616]
0x00007ff85e3ce428 UE4Editor-Core.dll!FWindowsErrorOutputDevice::Serialize() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Windows\WindowsErrorOutputDevice.cpp:78]
0x00007ff85e0df31d UE4Editor-Core.dll!FOutputDevice::LogfImpl() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Misc\OutputDevice.cpp:61]
0x00007ff85e075b55 UE4Editor-Core.dll!AssertFailedImplV() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Misc\AssertionMacros.cpp:104]
0x00007ff85e077bf0 UE4Editor-Core.dll!FDebug::CheckVerifyFailedImpl() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Misc\AssertionMacros.cpp:461]
0x00007ff8437397bf UE4Editor-D3D11RHI.dll!FD3D11DynamicRHI::RHIUpdateTextureReference() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Windows\D3D11RHI\Private\D3D11Texture.cpp:2260]
0x00007ff85f70a6a2 UE4Editor-RHI.dll!FRHICommandListImmediate::UpdateTextureReference() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\RHI\Private\RHICommandList.cpp:2730]
0x00007ff84d86dac1 UE4Editor-Engine.dll!FStreamableTextureResource::InitRHI() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Engine\Private\Rendering\StreamableTextureResource.cpp:183]
0x00007ff8600632cb UE4Editor-RenderCore.dll!FRenderResource::InitResource() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\RenderCore\Private\RenderResource.cpp:102]
0x00007ff8600bc65f UE4Editor-RenderCore.dll!TEnqueueUniqueRenderCommandType<BeginInitResource'::2’::InitCommandName,<lambda_62eddc873155d85469e17b1ba3893b43> >::DoTask() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\RenderCore\Public\RenderingThread.h:183]
0x00007ff8600cd4b0 UE4Editor-RenderCore.dll!TGraphTask<TEnqueueUniqueRenderCommandType<BeginInitResource'::2’::InitCommandName,<lambda_62eddc873155d85469e17b1ba3893b43> > >::ExecuteTask() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Public\Async\TaskGraphInterfaces.h:886]
0x00007ff85de8e3b4 UE4Editor-Core.dll!FNamedTaskThread::ProcessTasksNamedThread() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:709]
0x00007ff85de8e7be UE4Editor-Core.dll!FNamedTaskThread::ProcessTasksUntilQuit() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Async\TaskGraph.cpp:601]
0x00007ff8600eb532 UE4Editor-RenderCore.dll!RenderingThreadMain() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp:373]
0x00007ff8600f3c04 UE4Editor-RenderCore.dll!FRenderingThread::Run() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\RenderCore\Private\RenderingThread.cpp:509]
0x00007ff85e3e69bb UE4Editor-Core.dll!FRunnableThreadWin::Run() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:86]
0x00007ff85e3df680 UE4Editor-Core.dll!FRunnableThreadWin::GuardedRun() [C:\Users\123\Downloads\ue4\Engine\Source\Runtime\Core\Private\Windows\WindowsRunnableThread.cpp:35]
0x00007ff8c3477614 KERNEL32.DLL!UnknownFunction []
0x00007ff8c45026a1 ntdll.dll!UnknownFunction []

1. I have tried rebuilding the solution multiple times with different options, including the "UE4" option.
2. I installed the latest NVIDIA graphics driver (studio version).

Any help would be much appreciated. Thanks in advance.

ushio2021, 2023-01-19:
Same problem, same exception. Help needed.

doom6461, 2023-01-21:
Not sure if this helps your problem but I downgraded to 0.9.9 with UE_4.24 (installed from epic games launcher) and its working fine.

wx983, 2023-02-02:
> Thanks but I had no problems from 0.9.13 source on the same machine

Have you solved it ever? I encount the same issue now:
Assertion failed: ResourceTableFrameCounter == INDEX NONE[File:D:/UnrealEngine 4.26/Engine/Source/Runtime/Windows/D3D11RHI/Private/D3D11Texture.cpp] [Line: 2260].
I'm using carla 0.9.13 build from source.

ushio2021, 2023-02-05:
> Have you solved it ever? I encount the same issue now: **Assertion failed: ResourceTableFrameCounter == [...]

Hi, unfortunately I only came across this exception until 0.9.14.

ushio2021, 2023-05-30:
It seems the problem fixed (both DX11 and DX12) by upgrading to Windows Server 2022

andrejfsantos4, 2023-07-06:
I faced the same problem on CARLA 0.9.14 built from source in Windows and couldn't find any solutions, despite the issue being reported in multiple places ([link](https://forums.unrealengine.com/t/ue-4-26-built-from-source-crashes-when-opening-project/739873), #6201).

A hack I'm using that appears to work (so far at least) is to just comment out the line (l. 2260) that throws the error in D3D11Texture.cpp. Doing this I can now create projects and have seen no issues so far. 

But again, this is a hack and not a solution. It would be great to get an actual solution.

ushio2021, 2023-07-06:
> I faced the same problem on CARLA 0.9.14 built from source in Windows and couldn't find any solutions, despite the issue being reported in multiple places [...]

Although still not sure which one, I highly suspect that it is due to some Windows compablility issues. I no longer have these issues since my last reply above.

Gululu2231, 2024-01-01:
> I faced the same problem on CARLA 0.9.14 built from source in Windows and couldn't find any solutions, despite the issue being reported in multiple places [...]

> I faced the same problem on CARLA 0.9.14 built from source in Windows and couldn't find any solutions, despite the issue being reported in multiple places [...]

it works for me.  just comment out the line (l. 2260) that throws the error in D3D11Texture.cpp.

MTeuberItemis, 2024-01-15:
> it works for me. just comment out the line (l. 2260) that throws the error in D3D11Texture.cpp.

Hi, the error still persists, even after I comment out the line. It gives me the same error, in the same line, even though the line shouldnt actually exist.

MTeuberItemis, 2024-01-17:
Unfornately, the error persists even after I comment out the line. This makes no sense at all

prabanjanKAN, 2024-02-11:
Anyone found the solution for this? facing the same error

InigoAgirre, 2024-02-13:
Hello, I am having the same issue. Did someone manage to solve it?

prabanjanKAN, 2024-02-13:
Launching the UE4Editor.exe file with -dx12 argument (like: UnrealEngine\Engine\Binaries\Win64> .\UE4Editor.exe -dx12) resolved  this error for me in windows 11. 

If -dx12 argument doesn't work, try -dx11 or -vulkan or -opengl.

Bniultu, 2024-03-20:
> Launching the UE4Editor.exe file with -dx12 argument (like: UnrealEngine\Engine\Binaries\Win64> .\UE4Editor.exe -dx12) resolved this error for me in windows 11.
> 
> If -dx12 argument doesn't work, try -dx11 or -vulkan or -opengl.

This method helped me solve the problem by adding a -dx12 to the file path, I tried rebuild and turning off the graphics driver, etc., but the problem was not solved until I tried this method

evoevo07, 2024-03-24:
> This method helped me solve the problem by adding a -dx12 to the file path, I tried rebuild and turning off the graphics driver, etc., but the problem was not solved until I tried this method

Was your Carla make PythonAPI and make launch successful afterwards?

Bniultu, 2024-04-02:
> > > Launching the UE4Editor.exe file with -dx12 argument (like: UnrealEngine\Engine\Binaries\Win64> .\UE4Editor.exe -dx12) resolved this error for me in windows 11.
> > > If -dx12 argument doesn't work, try -dx11 or -vulkan or -opengl.
> > 
> > 
> > This method helped me solve the problem by adding a -dx12 to the file path, I tried rebuild and turning off the graphics driver, etc., but the problem was not solved until I tried this method
> 
> Was your Carla make PythonAPI and make launch successful afterwards? I am aware that this could be the solution for creating a project, but I suspect that the make process might fail

Did your Carla make PythonAPI and make succeed after launching?
I realize that this may be the solution to create the project, but I suspect that the make process may be missing

Yes, I've run both successfully and I've had no problems with the rest of the process.

Blyron, 2024-07-12:
Any news on this have your problems been fixed?

closed 2024-08-02 by Blyron

cuihangmver, 2024-12-25:
> -dx12

this method work for me, win10 vs2019 build for carla eu4