carla-simulator/carla #6077: Teleportation of pedestrian when spawning large number of pedestrian

LeoHuang0511, 2023-01-03:
CARLA version: 0.9.13
Platform/OS: Ubuntu 22.04.1 LTS
GPU: GeForce RTX 2080 Ti

I am trying to spawn crowd scene so I spawned lots of pedestrians in a small place with their bounding boxes and used a RGB camera to save the frames . However, some pedestrian teleport to somewhere next to them in several frames and get back to the normal trajectories. And when the problem occurs, the bounding boxes are still follow the normal trajectories. 
I used synchronous mode and a "while True" loop to save the frames. I've tried to add "time.sleep()" at the end of each loop, but it did't work.
I'm wondering that if there's any solutions can solve this problem, or it's just due to the GPU?
Thanks a lot.

TR-Monash, 2023-01-04:
I have observed the same when the number of pedestrians is around 100. I tested this on RTX 3080 Ti as well and issue is still there. So, I don't think it's due to the GPU. I don't have any solution for this problem either.

LeoHuang0511, 2023-01-05:
@TR-Monash I see. Did your problem occur when the collision happened (on two or more pedestrians) as well? I’m trying to use manual controller to avoid the problems. If you figure out any solutions, give me a hand, plz!

TR-Monash, 2023-01-05:
@LeoHuang0511 Not really. I mainly observed this when the ego vehicle is near a pedestrian. For example, a pedestrian walking across the path of the vehicle in a distance less than 2 metres (i'm trying to simulate near-accident scenarios). 
As of now, I don't really have any solution and I'm not even sure what's causing this behaviour, unfortunately.

lfxx, 2024-01-06:
same issue here.any updates on thi?