carla-simulator/carla #6085: Camera output distorted

AlvaroRausell, 2023-01-05:


MatthiasWildeBusiness, 2023-01-11:
Hi, 
I discovered a similar problem, which might be also your problem. When I changed my camera resolution to a aligned value (I used multiple of 64 pixels) I could solve it. It was needed to align for both, width and height:

I used width = 600  ==> stride problem
Align width to 640 ==> OK

Hope this helps you also.

vadim7s, 2023-01-12:
I have the same camera distortion problem after going to v 0.9.14. Then I went back to v0.9.13, everything works again. To test quickly you can run visualise_multiple_sensors.py in the examples folder in v13 vs.14. Does this mean it is v14 issue?

mawistc, 2023-01-12:
> I have the same camera distortion problem after going to v 0.9.14. Then I went back to v0.9.13, everything works again. To test quickly you can run visualise_multiple_sensors.py in the examples folder in v13 vs.14. Does this mean it is v14 issue?

And is your resolution aligned?

bernatx, 2023-01-13:
The problem happens only in Windows (DirectX) for resolutions not aligned to 256.
DirectX add some extra bytes to each row.
We have made a PR to fix that.

kevintsq, 2023-01-17:
Will you be releasing a version of this bug fix in the near future, as it is indeed a serious bug?

kevintsq, 2023-01-17:
Thank you, then I’ll wait for your new build instead of building my own version or reverting to the old one:)

TzabarDolev, 2023-09-16:
I just wrote a short blog post about this issue on my Carla research blog - https://carlasimblog.wordpress.com/2023/09/16/the-64-division-curse/. And added a short code snippet to fix this.

Noubissie, 2023-10-29:
[https://github.com/carla-simulator/carla/issues/6565#issuecomment-1722974703]this may be of help

SExpert12, 2024-05-24:
Solution for this given here:
https://github.com/TzabarDolev/carlablog/blob/main/fix_image_distortion.py

Blyron, 2024-07-12:
Is this problem fixed now?

closed 2024-07-26 by Blyron