carla-simulator/carla #6086: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000

DeLeonOscar, 2023-01-05:
Hello, 

I am using CARLA 0.9.13.

CARLA works when running a code for the first time. However, if I want to run the same code for the second time or any other code, a issue appears:

RuntimeError: time-out of 5000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:2000
[image]

Computer needs to be restarted in order to fix this issue. After I restart the computer, CARLA works fine just one time and continue the same error again. 

What could be the problem?

Kenneth71-P, 2023-01-25:
add port = 2000 just below import time

nusstu0019, 2023-10-06:
have you solved this problem?

Blyron, 2026-11-12:
The usual cause is that the first script left the world in synchronous mode when it exited, so the server waits for a tick that never comes. Restore the original settings when your script ends (`world.apply_settings(original_settings)`, also on exceptions). See [Synchrony and time-step](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_synchrony.md). Closing.

closed 2026-11-12 by Blyron