carla-simulator/carla #6088: Roadrunner to Carla unreal engine : Traffic island not imported

Moneyboy63211, 2023-01-06:


jimmyw404, 2023-01-06:
I noticed this also. Check out the fbx that RoadRunner generates, it does not include the Traffic Island. If you look at the MathWorks documentation it states that the Traffic Island tool import/exports into OpenDRIVE.

Moneyboy63211, 2023-01-06:
Facing issues when including traffic island in roadrunner. When export from roadrunner then import to Carla , the island doesn't show up.    Please advise. Thank you 

I need traffic island to create the scene as shown in the photo. Any alternatives ways?

[https://ibb.co/zPBCXKh](url)
[https://ibb.co/zFHvNcY](url)
[https://ibb.co/tD8Lz2f](url)

Moneyboy63211, 2023-01-06:
> 

I need traffic island to create the scene as shown in the photo. Any alternatives ways?

Blyron, 2026-11-12:
The traffic island is missing because RoadRunner's FBX export does not include it; it only exists in the OpenDRIVE file. Model the island as a mesh in RoadRunner, or add it in the editor, before importing. Closing, as this is a RoadRunner export limitation and not a CARLA bug.

closed 2026-11-12 by Blyron