carla-simulator/carla #6089: Towns 11 and 12 are empty

ricardodeazambuja, 2023-01-06:
Hi,

I just downloaded from the releases `CARLA_0.9.14.tar.gz` and `AdditionalMaps_0.9.14.tar.gz`. Then I copied opied the `AdditionalMaps_0.9.14.tar.gz` to the `Import` folder, ran `ImportAssets.sh` and launched the simulator. It automatically loads `Town10HD_Opt`, but when I ask it to load `Town11` or `Town12` it shows an empty world (only the sky and water below).

I tried it directly on my computer (Ubuntu 20.04, nvidia rtx 2080) and also using docker. So I am pretty sure the simulator is working because the other towns are loading without problems.

If you need any extra info, please, let me know.

ricardodeazambuja, 2023-01-06:
Ok, I got an answer on LinkedIn and I am pasting here:
`The large maps are loaded in tiles, so only the required assets are loaded into graphics memory (there's not enough space even on a big graphics card for the whole map) so the tiles near the ego vehicle are loaded. You need to spawn the ego vehicle to load the tiles. Run the manual_control.py script and the map should load.`
Is there another way to tell the engine to load the tiles without the ego vehicle? I am currently using it with pedestrians, cars, but "flying sensors" instead of the ego vehicle :smile:

glopezdiest, 2023-01-23:
No, all the tiles are loaded with respect to the ego vehicle's position. In any case, the "ego vehicle" is just an actor with its *rolename* attribute set to *hero* (like the manual control [here](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/manual_control.py#L248)), but it doesn't have to be a vehicle. As long as there is ONE AND ONLY ONE actor with the *hero* rolename, the large maps will work just fine

PD: We just recently added to the *dev* branch that the spectator also loads the map.

ricardodeazambuja, 2023-01-23:
@glopezdiest, thanks for the explanations!
I just tried the generate_traffic.py, with one vehicle and hero enabled, and I could use my [flying spectator](https://github.com/ricardodeazambuja/carla-simulator-python/blob/main/my_modified_files/flying_spectator_display.py) to fly all over Town12 without any problems. As soon as I removed the vehicle all the tiles were gone. 
I also tried to set the "role_name" of the RGB camera (Spectator with a RGB camera attached) to "hero", but the system doesn't show any tiles. I just pulled carlasim/carla:0.9.14 from docker hub.

closed 2023-01-27 by glopezdiest

ricardodeazambuja, 2023-01-27:
@glopezdiest, I would say my solution is more of a hack than a solution, but it's good enough for me :smile:. I don't mind having a vehicle doing nothing somewhere in the map. I realized I don't even need the generate_traffic.py running, I just need to create the actor (hero vehicle) and as long as I don't remove it, the tiles are generated for my spectator with sensors attached. Thanks for your support!