carla-simulator/carla #6092: Fatal Error during make launch

SinuThiru66, 2023-01-08:
I did everything according to the steps described in windows build 

I get the fatal error in make launch (used Visual Studio 2019 Community Edition) and when I analyze this error, it throws me this
[image]

The error is exception: read access violation RoadPainter.BluePrint.Object was nullptr, can anyone help and tell me what I can do 

Thanks!

bernatx, 2023-01-09:
Hi,
Have you checked if you have all the content files, in special the required file: **carla\Unreal\CarlaUE4\Content\Carla\Static\Decals\Road\RoadDirt\DI_RoadDirt_01.uasset**
Regards,
Bernat

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron