carla-simulator/carla #6094: Make pythonapi erron

Nighttell, 2023-01-09:
CARLA version:[0.9.14]
Platform/OS:Windows - 10.0.19044 - AMD64
Problem you have experienced:
make pythonapi erron

   -[install_zlib]: zlib has been successfully installed in "E:\GraduationDesign\carla\Build\zlib-install\"!
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "E:\GraduationDesign\carla\Build\" --zlib-install-dir "E:\GraduationDesign\carla\Build\zlib-install\"
    -[install_libpng]: Retrieving libpng.
    -[install_libpng]: Extracting libpng from "libpng-1.2.37-src.zip".
New-Object : 使用“3”个参数调用“.ctor”时发生异常:“找不到中央目录结尾记录。”
所在位置 C:\Windows\system32\WindowsPowerShell\v1.0\Modules\Microsoft.PowerShell.Archive\Microsoft.PowerShell.Archive.psm1:934 字符: 23
+ ... ipArchive = New-Object -TypeName System.IO.Compression.ZipArchive -Ar ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [New-Object]，MethodInvocationException
    + FullyQualifiedErrorId : ConstructorInvokedThrowException,Microsoft.PowerShell.Commands.NewObjectCommand

    -[install_libpng]: Removing "libpng-1.2.37-src.zip"
    -[install_libpng]: Removing dir "E:\GraduationDesign\carla\Build\manifest"
系统找不到指定的文件。
系统找不到指定的文件。
    -[install_libpng]: Creating "E:\GraduationDesign\carla\Build\libpng-1.2.37-source\build"
*.c
c1xx: fatal error C1083: 无法打开源文件: “E:\GraduationDesign\carla\Build\libpng-1.2.37-source\libpng\1.2.37\libpng-1.2.37-src\*.c”: No such file or directory
    -[install_libpng]: Creating "E:\GraduationDesign\carla\Build\libpng-1.2.37-install\lib"
    -[install_libpng]: Creating "E:\GraduationDesign\carla\Build\libpng-1.2.37-install\include"
系统找不到指定的路径。
系统找不到指定的路径。

CMake Error at CMakeLists.txt:2 (project):
  Generator

    Visual Studio 16 2019

  could not find any instance of Visual Studio.

-- Configuring incomplete, errors occurred!

this is cmd show
i have read document and can't find similar issue,and my vs is 2022,and i really need help

brscholz, 2023-01-09:
You need to install Visual Studio 2019 und launch "x64 Native Tools Command Prompt". Use this instead of cmd for the make commands.

Nighttell, 2023-01-09:
thanks，I try to take Visual Studio 16 2019 to Visual Studio 17 2022 in all bat files,and questions solved

closed 2023-01-09 by Nighttell