carla-simulator/carla #6096: Large map import failing

brscholz, 2023-01-09:
Hi everyone,

I created a large map in RoadRunner and followed the instructions given in the docs.
So I end up with 48 tiles, an .xodr and the following json file:

```
{
  "maps": [
      {
        "name": "A1_Vilnius_Kaunas_Endstueck",
        "xodr": "./A1_Vilnius_Kaunas_Endstueck.xodr",
        "use_carla_materials": true,
        "tile_size": 2000,
        "tiles": [ 
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_0_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_1_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_2_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_3_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_4_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_5_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_6_5.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_0.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_1.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_2.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_3.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_4.fbx",
        "./A1_Vilnius_Kaunas_Endstueck_Tile_7_5.fbx"
        ]
      }
  ],
  "props": []
}
```
I run make import ARGS="package=A1_Vilnius_Kaunas_Endstueck" and the process fails after a lot of busy time with the error:

```
Traceback (most recent call last):
  File "D:/carla/Util/BuildTools/Import.py", line 637, in <module>
    main()
  File "D:/carla/Util/BuildTools/Import.py", line 634, in main
    import_assets_from_json_list(json_list, args.batch_size)
  File "D:/carla/Util/BuildTools/Import.py", line 466, in import_assets_from_json_list
    load_asset_materials_commandlet(package_name)
  File "D:/carla/Util/BuildTools/Import.py", line 474, in load_asset_materials_commandlet
    invoke_commandlet(commandlet_name, commandlet_arguments)
  File "D:/carla/Util/BuildTools/Import.py", line 211, in invoke_commandlet
    subprocess.check_call(command, shell=True)
  File "C:\Users\rscholz\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 363, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['D:/UnrealEngine/Engine/Binaries/Win64/UE4Editor', 'D:\\carla\\Unreal\\CarlaUE4\\CarlaUE4.uproject', '-run=LoadAssetMaterials', '-PackageName=A1_Vilnius_Kaunas_Endstueck']' returned non-zero exit status 3.
make: *** [import] Fehler 1
```

Nevertheless, the map is available in the content folder. It is empty and the map origin z value is about -27000. I suppose that is due to the failure in when loading the asset materials, but it might be an independent error.

Hints or help would be very much appreciated.

Best, Robert

brscholz, 2023-01-09:
@bernatx, I think you're the large map god, aren't you?

closed 2023-01-09 by brscholz

FBanani, 2023-06-29:
> EDIT: A non-geolocated version of my map was working, so I only need to find out what's wrong with the geolocation.
> 
> Hi everyone,
> 
> I created a large map in RoadRunner and followed the instructions given in the docs. So I end up with 48 tiles, an .xodr and the following json file:
> 
> ```
> {
>   "maps": [
>       {
>         "name": "A1_Vilnius_Kaunas_Endstueck",
>         "xodr": "./A1_Vilnius_Kaunas_Endstueck.xodr",
>         "use_carla_materials": true,
>         "tile_size": 2000,
>         "tiles": [ 
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_0_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_1_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_2_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_3_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_4_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_5_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_6_5.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_0.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_1.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_2.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_3.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_4.fbx",
>         "./A1_Vilnius_Kaunas_Endstueck_Tile_7_5.fbx"
>         ]
>       }
>   ],
>   "props": []
> }
> ```
> 
> I run _make import ARGS="--package=A1_Vilnius_Kaunas_Endstueck"_ and the process finishes without error. The map is available in the content folder. It is empty and the map origin z value is about -27000.
> 
> [image]
> 
> Hints or help would be very much appreciated.
> 
> Best, Robert

Hi
Did you solve it?
I have this issue. Could you help me?