carla-simulator/carla #6099: Carla 0.9.14 Docker Image

i14d4, 2023-01-10:
Hello CARLA team,
thanks for the latest CARLA release with the new features and bug fixes !
I am really looking forward to use the 0.9.14 release but unfortunately I was not able to find the carla 0.9.14 docker image at dockerhub (https://hub.docker.com/r/carlasim/carla/tags). 

So I am wondering if there will be an official image released soon?

Thanks in advance and all the best!

hh0rva1h, 2023-01-13:
@bernatx Could you please release the docker image of 0.9.14 as well? Possibly even addressing https://github.com/carla-simulator/carla/issues/4900 while at it?

bernatx, 2023-01-13:
Hi, sure, I will publish it ASAP, with full maps :)

bernatx, 2023-01-13:
Done

i14d4, 2023-01-16:
@hh0rva1h  and @bernatx : Thanks a lot !

xczhanjun, 2023-01-19:
> Hi, sure, I will publish it ASAP, with full maps :)

great