carla-simulator/carla #6100: make pythonapi erron

Nighttell, 2023-01-10:
CARLA version:[0.9.14]
Platform/OS:Windows - 10.0.19044 - AMD64
Problem you have experienced:
make pythonapi erron
i want to know how to solve the problem in "make pythonapi"  in win10 x64 ,CARLA version:[0.9.14],and i find the similiar question https://github.com/carla-simulator/carla/issues/3076
but nobody answered,I really need help and i have taken two days installing carla,meeting so many bugs,i think maybe this is the last one  

-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "E:\GraduationDesign\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 168, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 121, in get_libcarla_extensions
    libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
FileNotFoundError: [WinError 3] 系统找不到指定的路径。: 'dependencies/lib'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "E:\GraduationDesign\carla\PythonAPI\carla\dist"!

Lijingqi97, 2023-03-10:
you can delete the file "carla\PythonAPI\carla\dependencies" and make PythonAPI again

yu-ju-chiu, 2023-05-11:
Do you solve the issue?
I met the issue and couldn't solve it.

TroyZhangFawang, 2024-08-18:
how to solve this problem? please

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron