carla-simulator/carla #6101: CarlaUE4.sh crash without logs in docker

YaoWang-Pony, 2023-01-10:


Pyrestone, 2023-10-09:
still having the same issue using a 20.04 container.

ralwing, 2025-03-03:
Currently, I'm facing the same error on Carla 0.9.15:
```
carla@0524aca6dfc2:~$ ./CarlaUE4.sh -nosound -RenderOffScreen 
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
carla@0524aca6dfc2:~$ ./CarlaUE4.sh -nosound -RenderOffScreen -opengl
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
carla@0524aca6dfc2:~$ nvidia-smi 
Mon Mar  3 08:36:01 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.120                Driver Version: 550.120        CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA TITAN RTX               Off |   00000000:03:00.0 Off |                  N/A |
| 41%   27C    P8              8W /  280W |       6MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   1  NVIDIA TITAN RTX               Off |   00000000:21:00.0 Off |                  N/A |
| 40%   27C    P8             14W /  280W |      16MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
|   2  NVIDIA TITAN RTX               Off |   00000000:49:00.0 Off |                  N/A |
| 40%   26C    P8              4W /  280W |       6MiB /  24576MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
+-----------------------------------------------------------------------------------------+

```

ralwing, 2025-03-03:
This might be related to:
https://github.com/carla-simulator/carla/issues/5743

Ridhamz-nd, 2025-07-07:
Hello @YaoWang-Pony , did you ever end up solving the issue? I tried looking into this with strace but I couldn't find anything yet.

Blyron, 2026-11-12:
This was a custom Docker image of CARLA 0.9.13 on Unreal Engine 4. Docker for CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) is documented in [Running CARLA in Docker](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_docker.md). Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron