carla-simulator/carla #6103: Default `gamma` value for the RGB/DVS camera has changed in version 0.9.14

vbrebion, 2023-01-11:
Hi!

Following the release of the 0.9.14 version of CARLA, I have noted that the default `gamma` setting for the RGB (and DVS) camera has changed, from a value of `2.2` to a value of `2.4`.

I don't know if this change was intended, as the documentation (<https://carla.readthedocs.io/en/0.9.14/ref_sensors/#rgb-camera>) still indicates that the default value is `2.2`. Furthermore, this value of `2.4` results in highly noisy DVS data at night, while `2.2` appears more realistic (see below).

Would it be possible for you to take a look at this issue?

Best,
Vincent

---

**CARLA versions:**
0.9.13 / 0.9.14

**Platform/OS:**
Ubuntu 22.04

**Problem you have experienced:**
As described above, the default `gamma` value seems to have changed.\
This change is especially visible with the DVS camera for a night scene:
- in version 0.9.13:
  [image]\
  [image]

- in version 0.9.14 with the default settings:
  [image]\
  [image]

- in version 0.9.14 after setting `gamma` to `2.2`:
  [image]\
  [image]

**Steps to reproduce:**
- Load the `sensor.camera.rgb` or the `sensor.camera.dvs` blueprint with default values
- Print its `gamma` attribute (should output `2.2` in version 0.9.13, `2.4` in version 0.9.14)

**Other information (documentation you consulted, workarounds you tried):**
/

Madecu, 2023-01-16:
Hi @vbrebion,

I'll check that out and see what happened with the gamma value.

Thanks for your report

Madecu, 2023-01-18:
Hi @vbrebion,

We have restored the default value to 2.2. This has been merge in dev and it will be in the next build.

Thank your for reporting the issue.

PR #6124 (2023-01-18): Restore gamma value to 2.2 (instead of 2.4)
#### Description

Restoring gamma value to 2.2

Fixes #6103

#### Where has this been tested?

  * **Platform(s):** windows
  * **Python version(s):** 3.7
  * **Unreal Engine version(s):** 4.26

#### Possible Drawbacks

- - -
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/6124)
closed 2023-01-18 by Madecu