carla-simulator/carla #6116: Carla 0.9.14 Windows Build error trying to use examples

mamaduke21, 2023-01-13:
Hi

I'm having a problem running generate_traffic.py. I follow every step of the Windows build and when i try to run an example always return an error. 
I notice that the API version is 0.9.12 but i don't know how to update to 0.9.14. Can someone help me please?

[image]

bernatx, 2023-01-18:
Hi, probably you have more than one file .egg into this folder:
`carla/PythonAPI/carla/dist/`
Check that you have only the .egg file for the 0.9.14 version

Another option is that you installed the .egg file. Then you need to uninstall that and install the new one.

mamaduke21, 2023-01-18:
Hi! 
Thank you for the help. 
Indeed there was more than one file .egg on that folder and it was causing the problem.

closed 2023-01-18 by bernatx