carla-simulator/carla #6119: Where to find the source-code of the External Sensor Interface (ESI)?

e-lab-carla, 2023-01-17:
Hello,
Our lab is currently trying to integrate more physically accurate Radar systems, with different modulations.
Therefor i was quite happy when i found out that the ESI did exist.
But i still haven't found the source code anywhere. Does anybody know what happened to the project, if it got discarded or is now available in a different name.

Thx

chrisochoatri, 2023-01-27:
+1 My understanding (from the discord) is that the ESI was halted due to funding issues. If the work-in-progress code is available, would love to take a look also.

repa1030, 2023-04-14:
Is there any update regarding funding and current development state of the external sensor interface?

Blyron, 2026-11-12:
The External Sensor Interface was never released. To add your own sensor, follow [Create a sensor](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/dev_create_sensor.md). Closing.

closed 2026-11-12 by Blyron