carla-simulator/carla #6120: CARLA 0.9.14 error by make launch - ""Renderer/Public/GBufferView.h" not found

LucasFKobernic, 2023-01-17:
CARLA version: 0.9.14
Platform/OS: Windows 10
Problem you have experienced:
By "make launch" I'm getting the following error:

[4/26] Module.Carla.gen.6_of_8.cpp
  C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\um\winnt.h(603): warning C4005: "TEXT": Makro-Neudefinition
  C:\UnrealEngine\Engine\Source\Runtime\Core\Public\HAL/Platform.h(1082): note: Siehe vorherige Definition von "TEXT"
  D:\carla_14\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(120): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_14\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Vehicle\WheeledVehicleMovementComponentNW.h(148): warning C4996: 'UVehicleWheel': PhysX is deprecated. Use the UChaosVehicleWheel from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_14\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(70): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_14\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(186): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla_14\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "Renderer/Public/GBufferView.h": No such file or directory

The same error appears more then one time, but always related with this "**Renderer/Public/GBufferView.h**". I checked that this is new (CARLA 0.9.13 didn't use it, right?).

Am i missing something new by compiling CARLA 0.9.14 ? Has anyone faced the same issue?

I took a look in the documentation, but found nothing that helped.

Thx!

closed 2023-01-17 by LucasFKobernic

fukuchi0105, 2023-01-27:
Hi
I got the same error. Is there any way to solve this?

Nash-dc, 2023-02-03:
It is probably that you use the old version of UE4.26.
Try to delete the old one then build the lastest CarlaUE4 version on your computer then build Carla

fukuchi0105, 2023-02-05:
Thank you for your comment! It works when I use the latest CarlaUE4.

cdefg, 2023-02-16:
Do you mean CarlaUE4's latest branch, or UnrealEngine 4.27+?
And I found the author "CarlaUnreal" has no public repository, how can I fetch the private repository of CarlaUE4?

Nash-dc, 2023-02-19:
I mean CarlaUE4's latest branch. 
In order to fetch the private repository of CarlaUE4, you need to follow the official instructions.
See: https://www.unrealengine.com/en-US/ue-on-github

cdefg, 2023-02-20:
I've already been a 'member' of UE, but I can't see any private repo in this link https://github.com/CarlaUnreal. Is this link right? Or should I find a specific branch from UE4 official repo.

Nash-dc, 2023-02-20:
This
https://github.com/CarlaUnreal/UnrealEngine.git

Nash-dc, 2023-02-20:
Fetch default branch

cdefg, 2023-02-20:
> Fetch default branch

Nice of you, hope you fine day!

cdefg, 2023-02-20:
I found my failure reason: I put the CarlaUE4 file in a long prefix path, and when it is compiling it ignores some items.
And now I put the file in D:\ root path, it has completely been built.