carla-simulator/carla #6121: Control Red light doesn't work

NnaYelsel, 2023-01-17:
CARLA version: 13
Platform/OS: Windows

Problem you have experienced: Change the time of the red lights doens't work

What you expected to happen: I used the information of ticket : [https://github.com/carla-simulator/carla/issues/515](https://github.com/carla-simulator/carla/issues/515) in order to modify the time of green, red and yellow lights.

- Green and yellow are working but not the red lights, even if the get methods gives the new values I put in. 
- The default value of 2s for red light is not the one I get when running the simulation without modifying the light.

Steps to reproduce:

    ```
for actors_ in list_actor:
        if isinstance(actors_, carla.TrafficLight):
            actors_.set_state(carla.TrafficLightState.Red)
            world.tick()
            actors_.set_red_time(1.0)
            world.tick()
            actors_.set_state(carla.TrafficLightState.Green)
            world.tick()
            actors_.set_green_time(3.0)
            world.tick()
            print(actors_.get_green_time())
            print(actors_.get_red_time())
            actors_.set_state(carla.TrafficLightState.Yellow)
            world.tick()
            actors_.set_yellow_time(0.1)
            world.tick()
            print(actors_.get_yellow_time())
```

I didn't find any bad coding in github nevertheless, all my red lights stay during nearly 10s whatever I do.

Best regards

bernatx, 2023-01-18:
Hi, we are going to check the traffic lights contgrollers

bernatx, 2023-01-18:
Hi,
I can not reproduce the problem. When I set the new values, the traffic lights are using the new times. 
But the red light is a bit special. Think in a junction with 4 traffic lights and two directions (a typical cross). 

- Two of the traffic lights are in group1
- and the others two are in group2. 

Each group run a cycle that goes from green state, to yellow state and ends with the red state. So using the default times, the group1 is 

- in green for 10 seconds
- then in yellow for 3 seconds
- and then in red for 2 seconds. 

Then what happens is that the cycle for the group1 ends, and the next group start a new cycle. That means that this group1 will remain in red state while the group2 is doing the cycle (that means group2 will be in red for 15 seconds). Then, when group2 ends the complete cycle (10 seconds in green, 3 seconds in yellow and 2 seconds in red), the group1 will start a new cycle of green + yellow + red, and so on. 
That means, that the group1 traffic lights will be in red (2 seconds during its cycle, and 15 seconds during the cycle of the other group) 17 seconds.

That is the normal behavior. Is that what you see?

CHNtentes, 2023-11-20:
@bernatx Hi,
You mentioned the traffic lights should work in pair at crossings, but in my simulation, only 1 of 4 lights is green at any time, the opposite light is still red. What could be wrong here?

axelbr, 2024-01-31:
I am experiencing the same behavior.

DnDare, 2024-05-17:
Hi, as mentioned in this [issue](https://github.com/carla-simulator/carla/issues/6932#issuecomment-2104463327), 
There's only one green traffic light active at the same time for each traffic light crossing.

seominseoktech, 2024-05-22:
@DnDare Hello. Then how can I accurately measure the time it takes for the red light to change?

DnDare, 2024-05-24:
Hi, the calculations you've made for 1 group traffic lights are correct, the full cicle lasts 17 seconds, 
But when there are more groups included, these times are added, as you mention at 
https://github.com/carla-simulator/carla/issues/7625#issuecomment-2124120231
 A single traffic light takes 15 seconds to make a full cicle,  10 green, 3 yellow, 2 red, then moves to a second traffic light, that starts with green light
 
 In Town10HD_Opt there are 3 different junctions or crossings with trafficlights, 
 the simplest takes 30 seconds for a whole loop, it's a simple junction of 2 trafficlights

A second junction combines 3 trafficLights, the whole loop takes 45 seconds

A third crossing combines 4 trafficLights, it takes 1 minute, 

You can review the information on trafficlights [here](https://carla.readthedocs.io/en/latest/python_api/#carlatrafficlight).

If you want to get the timing for each phase, get_green_time, get_yellow_time, get_red_time

I hope this helps you.

closed 2024-05-31 by Madecu