carla-simulator/carla #6125: ImportError: DLL load failed while importing libcarla

LucasFKobernic, 2023-01-17:
CARLA version: 0.9.14 and 0.9.13
Platform/OS: Windows 10
Problem you have experienced:
After sucessfully compiling the API (make PythonAPI), and CARLA (make launch). When running any python script I'm getting the following error:

```
Traceback (most recent call last):
  File "generate_traffic.py", line 24, in <module>
    import carla
  File "<frozen zipimport>", line 259, in load_module
  File "..\carla\dist\carla-0.9.14-py3.8-win-amd64.egg\carla\__init__.py", line 8, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "..\carla\dist\carla-0.9.14-py3.8-win-amd64.egg\carla\libcarla.py", line 9, in <module>
  File "..\carla\dist\carla-0.9.14-py3.8-win-amd64.egg\carla\libcarla.py", line 7, in __bootstrap__
ImportError: DLL load failed while importing libcarla: The specified module was not found.

```
The .egg file was created after running "make PythonAPI", and now error was displayed. By the error the problem seens to be with the .egg file. 

I'm facing this error by the versions 0.9.14 and 0.9.13. Interesting was, i had already 0.9.13 running, then i have recompiled it (make clean -> make pythonAPI) and got errors regarding the xerces library. after solving it by following the suggesting proposed here #3320, this new error appeared. 

Has anyone faced something similar or has any idea how to solve it?

Thx

bernatx, 2023-01-18:
Hi, to update to the 0.9.14 from the0.9.13, you need to pull also the latests changes in the forked UE4 and also get new content.
Also, version 0.9.14 uses new version of boost 1.80, so be sure that there is no conflict there.

suryachereddy, 2023-10-20:
I fixed the issue by not using Anaconda. Use venv instead.

stacyH8, 2024-06-25:
Hello,

My environment is :
      - Python 3.7
      - Carla 0.9.15
      - Windows 10
      - Python editor : Visual studio code

If you have this problem, it means that this python script doesn't locate in good place.
All the answers start from a good point by creating a virtual environment. Actually it should.

What I have tried and it can't work : 
Download zlib.dll, and put it under the path of Carla /PythonAPI. Nothing solved.

Actually for running well Carla in windows, it must create a virtual environment in python. In this environment, you can install all the necessary libraries to run Carla. It helps you avoid missing the packages of Carla when running it. I create a folder in disk C, and create my virtual environment in it.

How I solved this problem  <ImportError: DLL load failed while importing libcarla> : 
- My virtual environment is <sim-carla>, which is in a folder under disk C, absolutely all the necessary libraries to run Carla have already installed in this virtual environment. In this <sim-carla>, it has the folders : <Include>, <<lib>, <Scripts>, <Share> and <pyvenv.cfg>.
- The python script that I developed : I move it under a created folder. And this created folder is under the name of folder <sim-carla>. Then I execute the script without this error.

Hope that my solution is helpful. Good luck!

weizhii, 2025-01-11:
My Solution to the "ImportError: DLL load failed" Issue:
To resolve the import error stemming from the Carla library, I drew inspiration from (https://github.com/carla-simulator/carla/issues/1325) to develop my solution. I followed these steps:
envirment: 
python3.7.12  
win11 
carla 0.9.4

1.download zlib.dll x64
2.copy zlib.dll ,put zlib.dll under python\lib\site-package\carla-0.9.4-py3.7-win-amd64.egg\carla\ and  path of Carla /PythonAPI

Hope that my solution is helpful. Good luck!

closed 2025-04-23 by Blyron