carla-simulator/carla #6126: Assertion failed in CARLA client

jcontrolresearch, 2023-01-17:
CARLA version: 

- Client API version = 0.9.13 
- Simulator API version = 0.9.14

Platform/OS:

- Windows 10

Problem you have experienced:

Trying to follow the examples in PythonAPI/examples, I got the error in running the following scripts:

- dynamic_weather.py

Error:

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.13 
WARNING: Simulator API version  = 0.9.14 
A s s e r t i o n   f a i l e d :   p x   ! =   0 ,   f i l e   C : \ w o r k s p a c e \ c a r l a \ B u i l d \ b o o s t - 1 . 7 2 . 0 - i n s t a l l \ i n c l u d e \ b o o s t / s m a r t _ p t r / s h a r e d _ p t r . h p p ,   l i n e   7 2 8

bernatx, 2023-01-18:
Hi,
check this issue please
https://github.com/carla-simulator/carla/issues/6116

Pittawat2542, 2023-01-26:
I had encountered a similar issue and I found out that I need to upgrade my CARLA client library to match the version of CARLA server downloaded from GitHub Releases.

Be sure to upgrade the CARLA Python client library by using one of these options (excerpt from [official docs](https://carla.readthedocs.io/en/latest/start_quickstart/)):

1. CARLA provides .whl files for different Python versions. You will need to install the .whl file. The .whl file is found in `PythonAPI/carla/dist/`. There is one file per supported Python version, indicated by the file name (e.g., `carla-0.9.12-**cp36**-cp36m-manylinux_2_27_x86_64.whl` indicates Python 3.6).

```bash
 # Python 3
 pip3 install <wheel-file-name>.whl

 # Python 2
 pip install <wheel-file-name>.whl
```

If you previously installed the client library, you should [uninstall](https://carla.readthedocs.io/en/latest/build_faq/#how-do-i-uninstall-the-carla-client-library) the old one before installing the new one.

2. The CARLA client library can be downloaded from [PyPi](https://pypi.org/project/carla/). This library is compatible with Python versions 2.7, 3.6, 3.7, and 3.8. To install it you will need **pip/pip3** version 20.3 or above. See the [Before you begin](https://carla.readthedocs.io/en/latest/start_quickstart/#before-you-begin) section for how to check the version and upgrade **pip/pip3**.

**It is recommended to install the CARLA client library in a virtual environment to avoid conflicts when working with multiple versions.**

To install the client library from PyPi, run the following command:

```bash
 # Python 3
 pip3 install carla

 # Python 2
 pip install carla
```

The PyPi download is suitable for use with CARLA packages only (i.e., not with a version built from source). Since the PyPi download **only contains the client library**, it is most useful in situations where you will be communicating with a remote CARLA server where you do not require downloading a full CARLA package.

Hope it helps!

jcontrolresearch, 2023-01-26:
> I had encountered a similar issue and I found out that I need to upgrade my CARLA client library to match the version of CARLA server downloaded from GitHub Releases.
> 
> Be sure to upgrade the CARLA Python client library by using one of these options (excerpt from [official docs](https://carla.readthedocs.io/en/latest/start_quickstart/)):
> 
> 1. CARLA provides .whl files for different Python versions. You will need to install the .whl file. The .whl file is found in `PythonAPI/carla/dist/`. There is one file per supported Python version, indicated by the file name (e.g., `carla-0.9.12-**cp36**-cp36m-manylinux_2_27_x86_64.whl` indicates Python 3.6).
> 
> ```shell
>  # Python 3
>  pip3 install <wheel-file-name>.whl
> 
>  # Python 2
>  pip install <wheel-file-name>.whl
> ```
> 
> If you previously installed the client library, you should [uninstall](https://carla.readthedocs.io/en/latest/build_faq/#how-do-i-uninstall-the-carla-client-library) the old one before installing the new one.
> 
> 2. The CARLA client library can be downloaded from [PyPi](https://pypi.org/project/carla/). This library is compatible with Python versions 2.7, 3.6, 3.7, and 3.8. To install it you will need **pip/pip3** version 20.3 or above. See the [Before you begin](https://carla.readthedocs.io/en/latest/start_quickstart/#before-you-begin) section for how to check the version and upgrade **pip/pip3**.
> 
> **It is recommended to install the CARLA client library in a virtual environment to avoid conflicts when working with multiple versions.**
> 
> To install the client library from PyPi, run the following command:
> 
> ```shell
>  # Python 3
>  pip3 install carla
> 
>  # Python 2
>  pip install carla
> ```
> 
> The PyPi download is suitable for use with CARLA packages only (i.e., not with a version built from source). Since the PyPi download **only contains the client library**, it is most useful in situations where you will be communicating with a remote CARLA server where you do not require downloading a full CARLA package.
> 
> Hope it helps!

Thanks to share!

In fact, my issue was related with the python version that I had on my conda environment. The Python installed was 3.9 instead 3.8. The solution was creating a new environment with python 3.8 and install the .whl located in `PythonAPI/carla/dist/`
After that client examples runs without any issue.

kabdelmonem, 2024-03-04:
hello i am encountring the same problem using carla 9.15 but only for town 03 , the other towns work without problem , i have verified that the python version in my conda environment  is 3.7 and that the correct wheel is installed , the error is as follows : 
the town 3 map is loaded but no simulation is launched
Assertion failed: px != 0, file C:/Users/Public/Public_CARLA/CARLA_source_code/carla/Build/boost-1.80.0-install/include\boost/smart_ptr/shared_ptr.hpp, line 783

m-shahbaz-kharal, 2025-04-09:
> hello i am encountring the same problem using carla 9.15 but only for town 03 , the other towns work without problem , i have verified that the python version in my conda environment is 3.7 and that [...]

Same issue for me; it works fine on another map but this one I created it gives this exact same assertion error.

Blyron, 2025-04-09:
make sure you have a proper simulator version matching client version. This can be found in logs of client

closed 2025-07-27 by Blyron