carla-simulator/carla #6130: Memory leak in PixelReader.h

brscholz, 2023-01-19:
Hi guys,

today we noticed that we run out of memory when simulating with a camera.
A quick look with UnrealInsights showed that the memory tagged with EngineMisc is increasing steadily.

[image]

In my local version, I have the possibility to deactivate sending the data to python. i.e. to not execute the code in PixelReader.h.
When I use that option, the memory leak does not occur, so it seems like some problem with PixelReader.h.

Best Robert

MarcelPiNacy, 2023-02-22:
Hi Robert,

I've been looking into a similar problem for a while now.
Are you using synchronous mode from python by any chance?

Kind regards,
Marcel

brscholz, 2023-02-22:
Hi Marcel,

yes, I am. Exclusively that mode, tbh. :)

Best Robert

lygbuaa, 2024-02-18:
Hi, is this memory leak problem arised since 0.9.14 ?
Recently, I encountered memory leak in 0.9.14 and 0.9.15, but not in 0.9.13 with the same python scripts.

Blyron, 2025-02-19:
This problem should be solved now on latest ue4-dev branch