carla-simulator/carla #6133: How to apply auto blinker to my_vehicle and npc vehicles?

Kenneth71-P, 2023-01-20:
manual_control.py(PythonApi example file)

class KeyboardControl(object):
    """Class that handles keyboard input."""
    def __init__(self, world, start_in_autopilot):
        self._autopilot_enabled = start_in_autopilot
        if isinstance(world.player, carla.Vehicle):
            self._control = carla.VehicleControl()
            self._lights = carla.vehicle.set_light_state(carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam)
            world.player.set_autopilot(self._autopilot_enabled)
            world.player.set_light_state(self._lights)
        elif isinstance(world.player, carla.Walker):
            self._control = carla.WalkerControl()
            self._autopilot_enabled = False
            self._rotation = world.player.get_transform().rotation
        else:
            raise NotImplementedError("Actor type not supported")
        self._steer_cache = 0.0
        world.hud.notification("Press 'H' or '?' for help.", seconds=4.0)

I wrote "self._lights = carla.vehicle.set_light_state(carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam)" in manual_control.py>class KeyboardControl(object) because I want to apply auto blinker to my vehicle and npc vehicles when these vehicle are turning left or right and the vehicles' autopilot was enabled. Then I run manual_control.py file, but the cmd screen says module 'carla' has no attribute 'vehicle'. 
So can anyone tell me how to solve the problem?

My System:
Window 11
Python 3.7.9
Carla 0.9.13
Unreal engine file: I don't have it

[image]
[image]

glopezdiest, 2023-01-23:
Hey, you are getting confused between the *carla.Vehicle* function and the *carla.VehicleLightState* enum. First define the lights that you want to be turned on and off. This are independent from the vehicle so forget about the `carla.vehicle.set_light_state`. 

```
self._lights = carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam
```

After that you decide which vehicle will have these lights turned on. Now you call the `carla.Vehicle.set_light_state` function, and you are already doing this correctly at line 377:

```
world.player.set_light_state(self._lights)
```

As *world.player* is in this case, the manually controlled actor instance.

This works for all vehicles, but in the case of the autopilot, you can actually use the `carla.TrafficManager.update_vehicle_lights(world.player, True)` function to make the autopilot automatically change the vehicle's lights

Closing the issue but feel free to reopen it if you have any other related questions.

closed 2023-01-23 by glopezdiest

Kenneth71-P, 2023-01-23:
> Hey, you are getting confused between the _carla.Vehicle_ function and the _carla.VehicleLightState_ enum. First define the lights that you want to be turned on and off. This are independent from the vehicle so forget about the .`carla.vehicle.set_light_state`
> 
> ```
> self._lights = carla.VehicleLightState.RightBlinker | carla.VehicleLightState.LowBeam
> ```
> 
> After that you decide which vehicle will have these lights turned on. Now you call the function, and you are already doing this correctly at line 377:`carla.Vehicle.set_light_state`
> 
> ```
> world.player.set_light_state(self._lights)
> ```
> 
> As _world.player_ is in this case, the manually controlled actor instance.
> 
> This works for all vehicles, but in the case of the autopilot, you can actually use the function to make the autopilot automatically change the vehicle's lights`carla.TrafficManager.update_vehicle_lights(world.player, True)`
> 
> Closing the issue but feel free to reopen it if you have any other related questions.

Ok, thanks and I'll try it.

Kenneth71-P, 2023-01-27:
> Hey, you are getting confused between the _carla.Vehicle_ function and the _carla.VehicleLightState_ enum. First define the lights that you want to be turned on and off. This are independent from [...]

Sorry to bother you, but I have another question about it. Which python api should i put "carla.TrafficManager.update_vehicle_lights(world.player, True)" in?