carla-simulator/carla #6138: Update Documentation for Instance Segmentation Camera: Output does not match label ids specified in documentation

MarcoRiedenauer, 2023-01-22:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04
Problem you have experienced: The output of the instance segmentation camera does not use the same red-values as specified in the documentation. E.g. cars should have the red-value 10, but have 14 in the output image.
What you expected to happen: The documentation should contain the correct values for each class.

Please update the documentation (https://carla.readthedocs.io/en/latest/ref_sensors/#instance-segmentation-camera)

MeinLieben, 2023-01-30:
This problem is also in the Cityscapes palette. For example the fence class may have a different color than the color in the documentation ({153,153,90} if I remember it right). 
I suppose that this is not only throuble with the documentation cause there are similar classes in sources (for instance in CityScapesPalette.h or ObjectLabel.h).  So it looks more like a bug in the source code, but I haven't gone into it.

KevinLADLee, 2023-12-20:
https://github.com/carla-simulator/carla/blob/d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/LibCarla/source/carla/rpc/ObjectLabel.h#L16-L50

I also found this issue. And this part may help U to find out write label type ID. I think the documentation should be updated.

M-Colley, 2026-10-26:
PR #10010 has been merged into `ue4-dev` and corrects the instance-segmentation example to use the current Car label ID, 14 (10 is Terrain). This is the example/documentation correction; it does not claim a change to camera encoding or a general palette mismatch.

Blyron, 2026-11-12:
The documentation has been fixed: the instance segmentation section of [ref_sensors.md](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/ref_sensors.md) now matches the semantic tags (R = 14 is Car). Closing.

closed 2026-11-12 by Blyron