carla-simulator/carla #6139: after install ROS , and my working carla stop working

Moneyboy63211, 2023-01-23:


Moneyboy63211, 2023-01-23:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.9.14
Platform/OS: ubuntu 20.04
Problem you have experienced: so previosusly i got a working CARLA 0.9.13 but after o install ROS in my evironement, my CARLA stop working. so i reinstall whole carla again with the latest version of 0.9.14. But the error still persist even newly clone from github.

i got the following errror when "make PythonAPI"

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make PythonAPI
Using clang-10 as the CARLA compiler.
Setup.sh: Retrieving libc++.
Cloning into 'llvm-8.0-source'...
remote: Enumerating objects: 36751, done.
remote: Counting objects: 100% (36751/36751), done.
remote: Compressing objects: 100% (32058/32058), done.
remote: Total 36751 (delta 4686), reused 34943 (delta 4021), pack-reused 0
Receiving objects: 100% (36751/36751), 50.81 MiB | 15.88 MiB/s, done.
Resolving deltas: 100% (4686/4686), done.
Updating files: 100% (35358/35358), done.
Cloning into 'llvm-8.0-source/projects/libcxx'...
remote: Enumerating objects: 8052, done.
remote: Counting objects: 100% (8052/8052), done.
remote: Compressing objects: 100% (4726/4726), done.
remote: Total 8052 (delta 3946), reused 5825 (delta 3040), pack-reused 0
Receiving objects: 100% (8052/8052), 3.70 MiB | 11.88 MiB/s, done.
Resolving deltas: 100% (3946/3946), done.
Cloning into 'llvm-8.0-source/projects/libcxxabi'...
remote: Enumerating objects: 136, done.
remote: Counting objects: 100% (136/136), done.
remote: Compressing objects: 100% (129/129), done.
remote: Total 136 (delta 24), reused 28 (delta 2), pack-reused 0
Receiving objects: 100% (136/136), 743.04 KiB | 3.08 MiB/s, done.
Resolving deltas: 100% (24/24), done.
Setup.sh: Compiling libc++.
-- The C compiler identification is Clang 10.0.0
-- The CXX compiler identification is Clang 10.0.0
-- The ASM compiler identification is Clang
-- Found assembler: /usr/bin/clang-10
-- Check for working C compiler: /usr/bin/clang-10
-- Check for working C compiler: /usr/bin/clang-10 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++-10
-- Check for working CXX compiler: /usr/bin/clang++-10 -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++-10"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/tianle/Documents/carla/Build/llvm-8.0-build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/home/tianle/venv_carla/.venv/bin/ninja cmTC_70da7 && [1/2] Building CXX object CMakeFiles/cmTC_70da7.dir/testCXXCompiler.cxx.o
    [2/2] Linking CXX executable cmTC_70da7
    FAILED: cmTC_70da7 
    : && /usr/bin/clang++-10     CMakeFiles/cmTC_70da7.dir/testCXXCompiler.cxx.o  -o cmTC_70da7   && :
    /usr/bin/ld: cannot find -lstdc++
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:39 (project)

-- Configuring incomplete, errors occurred!
See also "/home/tianle/Documents/carla/Build/llvm-8.0-build/CMakeFiles/CMakeOutput.log".
See also "/home/tianle/Documents/carla/Build/llvm-8.0-build/CMakeFiles/CMakeError.log".
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

~~~~~~~~~~~~~~~

Blyron, 2023-01-23:
Hey I have read your post.

Please review you have lstdc++ correctly installed. The log it says is missing. This usually comes with Linux.

Thanks for using CARLA

Moneyboy63211, 2023-01-23:
> Hey I have read your post.
> 
> Please review you have lstdc++ correctly installed. The log it says is missing. This usually comes with Linux.
> 
> Thanks for using CARLA

hi thanks for helping me

i checked. the libstdc++ is installed, so i not sure what is the root cause 
[image]

Sid1057, 2023-01-28:
Hello @Moneyboy63211 
Could you describe: what happens when you try to run carla (as I see - you write an error when you try to build carla from source)
And what version of ROS are you using (humble/galactic/etc)

Moneyboy63211, 2023-01-28:
Hi

thank you for your reply.

I found that once I installed ROS noetic, my build of Carla from source
will stopped working. When "make import" or "make pythonAPI", the above
error occurs.

vttitta, 2023-02-20:
Hey

I'm having the exact same issue. I installed ros2 bridge to carla earlier today and now I'm trying to use the make Import and getting a cmake error. This happens with make PythonAPI too.

miaoxw, 2023-07-10:
I have the same issue. It happes on CARLA 0.9.14 with ROS noetic and Ubuntu 20.04.

closed 2024-05-24 by Blyron