carla-simulator/carla #6146: Creating a Realistic PhysX CARLA Vehicle

jimmyw404, 2023-01-24:
Hello CARLA team,

I'd like to create a CARLA vehicle whose vehicle physics are reasonably similar to its real-world model. I'm familiar with the instructions at https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/ but am specifically looking at how to parameterize the vehicle such that it makes the best use of the underlying PhysX model to provide realistic vehicle motion. I am familiar with the documentation and tuning guide at https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html#tuning-guide and https://docs.unrealengine.com/4.27/en-US/InteractiveExperiences/Vehicles/VehicleUserGuide/ that describes the UWheeledVehicleMovementComponent4W the CARLA vehicle's are built on, however I have not seen any resources on how to parameterize the PhysX configuration of a specific vehicle using data backed method like manufacturer specifications, measurements or vehicle recordings. We have experience with using CarSim and are exploring using CARLA without it (or Project Chrono) for this effort.

There is an impressive library of real-world vehicles within CARLA. I have a few specific questions on them:

* Is there any documentation on the process that can be shared on how those vehicles were configured to match their real-world counterparts?
* Are there any particular set of vehicles whose physics calibrations were made more realistic than others?
* Is there any data that can be shared on what was used to model the physics of a given vehicle?
* Are there any pitfalls you've run into when using PhysX that create an upper-boundary on vehicle realism?

@XGodina @DSantosO @jackbart94 @glopezdiest @joel-mb

jimmyw404, 2023-01-24:
Issue related to  #2847 https://github.com/carla-simulator/carla/issues/2847

ushio2021, 2023-02-05:
Really interested in this topic as well, also have read these documents but do not know how to achieve the same goal.

iasonastsaousis, 2025-04-02:
Any updates on this??

closed 2025-07-27 by Blyron