carla-simulator/carla #6151: Xerces-c-3.2.4 can't be downloaded from the website, although the file has existed in the website clearly and can be downloaded via the browser

wx983, 2023-01-25:
OS: Windows10
UnrelEngine: 4.26
Carla: 0.9.13
Visual Studio 2019 Community

**Means I have tried** :
1. Because 3.2.3 version of Xerces is no longer in the website: https://downloads.apache.org/xerces/c/3/sources, I have replaced 3.2.3 with 3.2.4 in \Util\BuildTools\Setup.bat, as well as in \Util\InstallersWin\install_xercesc.bat
2. download manually 3.2.4.zip in the browser to \Build folder
3. I have stable access to the  website: https://downloads.apache.org/xerces/c/3/sources

**My problem:** While buiding carla on windows using 'make PythonAPI' in x64 Native Tools cmd of VS 2019, **Xerces-c-3.2.4 can't be downloaded from the website, although the file has existed in the website clearly and can be downloaded via the browser**

**Error Report Information From x64 Native Tools cmd of VS 2019:**
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_xercesc]: Retrieving xerces-c.
    -[install_xercesc]: Using xerces-c from backup.

    -[install_xercesc]: [DOWNLOAD ERROR] An error ocurred while downloading xerces.
    -[install_xercesc]: [DOWNLOAD ERROR] Possible causes:
    -[install_xercesc]:              - Make sure that the following url is valid:
    -[install_xercesc]: "https://downloads.apache.org/xerces/c/3/sources/xerces-c-3.2.4.zip"
    -[install_xercesc]: [DOWNLOAD ERROR] Workaround:
    -[install_xercesc]:              - Download the xerces's source code and
    -[install_xercesc]:                extract the content in
    -[install_xercesc]:                "D:\carla\Build\xerces-c-3.2.4-source\"
    -[install_xercesc]:                And re-run the setup script.
    -[install_xercesc]: Exiting with error...

It has troubled me for four days, searching for the Internet and I even have tried to modify the bat file, but I can't still solve it.

wx983, 2023-01-28:
I have solved the issue. I used the conda environment before. But after abandoning it, 'make pythonAPI' successfully.

lygbuaa, 2023-01-29:
Try setting `XERCESC_REPO=https://archive.apache.org/dist//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz` in Util/BuildTools/Setup.sh

wx983, 2023-01-30:
> 尝试`XERCESC_REPO=https://archive.apache.org/dist//xerces/c/3/sources/xerces-c-${XERCESC_VERSION}.tar.gz`在 Util/BuildTools/Setup.sh 中设置

Thanks! I have aleady built carla successfully not using virtual conda environment, but ordinary python3.8.1. I uesd https://downloads.apache.org/xerces/c/3/sources/xerces-c-3.2.4.zip, it's OK!

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron