carla-simulator/carla #6157: Large Maps: SensorData.transform is in local (tile) coordinates

cscribano, 2023-01-26:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04

**Problem you have experienced:** 
When acquiring data from a sensor (i.e, camera) in a Large Map (Town12) the returned SensorData.transform is not in global map coordinates, instead a call to Sensor.get_transform() provides the correc transform.

**What you expected to happen:** 
The provided transform should be instead in World coordinates.

**Additional Information:**

The root of the problem is that "FAsyncDataStreamTmpl" (AsyncDataStream.cpp) obtains the transform trough AActor.GetActorTransform()

As a (dirty) proof-of-concept, I redefined this method in sensor.h
```
FTransform ASensor::GetActorTransform() const
{
  FTransform ActorTransform = Super::GetActorTransform();
  ALargeMapManager * LargeMap = UCarlaStatics::GetLargeMapManager(GetWorld());
  if (LargeMap)
  {
    ActorTransform = LargeMap->LocalToGlobalTransform(ActorTransform);
  }
  return ActorTransform;
}
```

This workaround did the trick!

However, I believe that the root cause of this, and potentially other analogous issues, should be further investigated by someone with a deeper knowledge.

glopezdiest, 2023-02-01:
Hey, We'll take a look at it but you are most likely right. We've found out that some of our transforms are lacking the large map transform, and this can definitely be one of those. Thanks for bringing this up :+1:

dmholtz, 2025-01-28:
CARLA version: 0.9.15 
Platform / OS: Ubuntu 22.04

The same issues seems to apply for bounding boxes for traffic light boxes, which are obtained via `get_light_boxes()` method of a TrafficLight instance. The issue only exists in large maps such as Town13.

@glopezdiest can you confirm whether this is the same cause? Are there any updates on whether a solution has been integrated in ue4-dev / previous master branch?

Blyron, 2026-11-12:
This was specific to the Unreal Engine 4 large-map tiling, where the world origin was rebased per tile. CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) uses World Partition and large world coordinates with no origin rebasing, so sensor transforms are in global coordinates. Closing as obsolete. If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron