carla-simulator/carla #6162: Low fps with low PC hw utilization on multiple sensor allocated.

dimitryn, 2023-01-30:
Hi ,

CARLA version:0.9.12
Platform/OS:Windows
Problem you have experienced: Low FPS 
Is there way to run simulator with 4 camera sensor and IMU sensor in Real Time?
I have CPU :Intel Core i7-8700k CPU 3.7Gz
           GPU: NVIDIA GeForce RTX 3090 
both client ans server are local.
I am using manual_contrtol.py example to simulate flow and I running in sync mode.
I am creating 5 sensor 1xIMU + 2xRGB + 2xDepth Sensor. my CPU is at 20% and GPU at 7% utilization. Memory at 12%. 
Unreal Editor showing 20-25 fps . 

What you expected to happen:
I would expect if I cannot run in real time is do to high usage of cpu or gpu or memory but here I see all resource in low utilization .

Any suggestion ? How do I know what is my bottleneck here ?
Thanks.

andrejfsantos4, 2024-04-23:
I'm experiencing a very similar issue on 0.9.15

Blyron, 2024-05-10:
Using multiple cameras at the same time is quite expensive due to we are sending data to the client secuencially. That's why you are having this performance. We are aware of it and we will try to improve this in the future

closed 2024-05-10 by Blyron

repa1030, 2024-12-11:
Any updates on that? Running the example/manual_control.py causes similar issues at my system. Even though i can run at around 15 fps, I expected higher values. Utilization of the hardware is ~50% CPU, ~11% GPU, ~20% RAM. What I sould mention here is that removed all sensors beside one camera and server and client running on the same machine.