carla-simulator/carla #6167: Boost `failed msvc.archive ...` error

thomiaditya, 2023-01-31:
CARLA version: 0.9.14
Platform/OS: Windows 10 64 bit
## Problem:

I tried to build carla dependencies using `make setup`, but always got an error on `boost` like this:

```sh
...
compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\scalars.obj
08:54:12:497	    scalars.cpp
08:54:12:508	    compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\ufunc.obj
08:54:12:509	    ufunc.cpp
08:54:12:608	    msvc.archive build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\libboost_numpy311-vc142-mt-x64-1_80.lib
08:54:12:608	    link: extra operand '/out:build\\boost\\bin.v2\\libs\\python\\build\\msvc-14.2\\release\\link-static\\python-3.11\\threading-multi\\libboost_numpy311-vc142-mt-x64-1_80.lib'
08:54:12:608	    Try 'link --help' for more information.
08:54:12:741	    
08:54:12:741	            if exist "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\libboost_numpy311-vc142-mt-x64-1_80.lib" DEL "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\libboost_numpy311-vc142-mt-x64-1_80.lib"
08:54:12:741	            call "build\boost\bin.v2\standalone\msvc\msvc-14.2\msvc-setup.bat"  >nul
08:54:12:741	     link /lib /NOLOGO  /out:"build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\libboost_numpy311-vc142-mt-x64-1_80.lib" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\dtype.obj" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\matrix.obj" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\ndarray.obj" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\numpy.obj" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\scalars.obj" "build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\numpy\ufunc.obj"   
08:54:12:742	        
08:54:12:742	    ...failed msvc.archive build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi\libboost_numpy311-vc142-mt-x64-1_80.lib...
08:54:12:742	    ...skipped <pG:\all-project\thesis-project\campus-av\dependencies\carla-source\Build\boost-1.80.0-install\lib>libboost_numpy311-vc142-mt-x64-1_80.lib for lack of <pbuild\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi>libboost_numpy311-vc142-mt-x64-1_80.lib...
08:54:12:742	    ...skipped <pbuild\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi>libboost_numpy-variant-vc142-mt-x64-1_80-static-py3.11.cmake for lack of <pbuild\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi>libboost_numpy311-vc142-mt-x64-1_80.lib...
08:54:12:742	    ...skipped <pG:\all-project\thesis-project\campus-av\dependencies\carla-source\Build\boost-1.80.0-install\lib\cmake\boost_numpy-1.80.0>libboost_numpy-variant-vc142-mt-x64-1_80-static-py3.11.cmake for lack of <pbuild\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.11\threading-multi>libboost_numpy-variant-vc142-mt-x64-1_80-static-py3.11.cmake...
08:54:12:742	    ...failed updating 6 targets...
08:54:12:742	    ...skipped 18 targets...
08:54:12:743	    ...updated 16584 targets...
08:54:12:776	    
08:54:12:777	        -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
08:54:59:463	        -[install_boost]: Exiting with error...
...
```

Tools that Im using to produce that errors:
- Visual Studio 2019 msvc 14.2
- CMake
- Make 3.81

After I analyze the errors after trying a lot of times with a lot of configurations (like changing using `Visual Studio 2022 msvc 14.3` and changing my cmake variables, also changing the directory), I found that it sometimes can be successfully built but in occasional moments, I don't know what I did wrong at that time.

ronyshaji, 2023-06-15:
Hey, @thomiaditya Did you find a work around?

Blyron, 2026-11-12:
This is an error from the old Make/`.bat` build of CARLA 0.9.x on Unreal Engine 4. That build system is gone: CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`) builds with CMake, which downloads and builds the dependencies itself. See [Build on Windows](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/build_windows_ue5.md). Closing as obsolete. If the current build fails for you, please open a new issue with the full log.

closed 2026-11-12 by Blyron