carla-simulator/carla #6172: 'make launch' failed after installing SRanipal and Logitech Wheel Plugin: assertion failed

wx983, 2023-02-02:
build carla 0.9.13 from source

I was able to use carla before installing SRanipal and Logitech Wheel Plugin.
I've already 'make pythonAPI' again after installing above Plugins and it reports no errors seemingly from the log information. But when 'make launch', it generates an error as beblow:
[image]

Dreyevr, 2023-03-20:
were you able to solve this error? I have the same problem.

wx983, 2023-03-21:
> 你能解决这个错误吗？我也有同样的问题。

Yes, try install UnrealEngine 4.26 and Carla0.9.13. At the same time, you must ensure the SRanipal Plugin is 1.3.3.0 rather than 1.3.6.+. 
I reinstalled them to solve the issue.

Blyron, 2026-11-12:
This report is about an old CARLA release on Unreal Engine 4, which is no longer maintained. Closing as obsolete. Please try CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`). If it still happens there, please open a new issue with the steps to reproduce and the logs.

closed 2026-11-12 by Blyron