carla-simulator/carla #6173: Town11 and Town12 cannot load

Miknow78, 2023-02-02:
I'm using the new realease 0.9.14 on ubuntu.
Everything works fine except the 2 new worlds: Town11 and Town12. 
It just loads the sky and I already tried to spawn a hero vehicle but the map still doesn't load. What do I do?
[image]

DnDare, 2023-02-02:
Hi, Town11 and Town12 are big maps, they are divided in sublevels that are loaded according the camera's or vehicle's  position.

try to hit play and execute a Manual_control.py 

If you want to find each sublevel
[image]

SExpert12, 2023-06-05:
Hi, my system configuration are:
12 GB GPU
Ubuntu 20.04 (focal)
Carla 0.9.14
I have included assets of Town11 and Town12.  Now every time I load map of Town11 and Town12 I get this and it got stuck.

How to access these towns?
I want to check intersection area in town11 and town12.
[image]

Alphacool123, 2023-06-06:
> Hi, my system configuration are: 12 GB GPU Ubuntu 20.04 (focal) Carla 0.9.14 I have included assets of Town11 and Town12. Now every time I load map of Town11 and Town12 I get this and it got stuck.
> 
> How to access these towns? I want to check intersection area in town11 and town12. [image]

I met the same question, do you fix it?

viola521, 2023-07-08:
how could fix it ? I meet  the same problem.

viola521, 2023-07-12:
> Manual_control.py

how could I open the configuration tool?

SExpert12, 2023-07-14:
I am getting your question? Can you please explain what you want to do?

Smarome, 2023-08-11:
> I'm using the new realease 0.9.14 on ubuntu. Everything works fine except the 2 new worlds: Town11 and Town12. It just loads the sky and I already tried to spawn a hero vehicle but the map still doesn't load. What do I do? [image]

Have you solved the problem? I can't open the map with 3090

SExpert12, 2023-08-14:
No. It's not getting loaded.

WenYanger, 2023-10-07:
The problems is due to the large size of Town11 and Town12 map.

The solution is to spawn an actor with role_name 'hero', then Carla will load the map tile around the actor.

For example,
      bp=random.choice(world.get_blueprint_library().filter('*vehicle*'))
      bp.set_attribute('role_name', 'hero')
      spawn_point = random.choice(world.get_map().get_spawn_points())
      ego = world.try_spawn_actor(bp, spawn_point)

SExpert12, 2023-10-10:
Okay. Thanks.

Blyron, 2024-05-10:
Is okay to get stucked for a while. The map needs to load heavily. 
Are you okay with this answer?

closed 2024-05-24 by Blyron

TomoCoder123, 2025-05-29:
Hello, I am on 9.15, and when I try spawning in a hero vehicle, it just spawns me under the map
I used this script.
`client = carla.Client('localhost', 2000)
client.set_timeout(60.0)  
world = client.load_world('Town12')

bp=random.choice(world.get_blueprint_library().filter('vehicle'))
bp.set_attribute('role_name', 'hero')
spawn_point = random.choice(world.get_map().get_spawn_points())
ego = world.try_spawn_actor(bp, spawn_point)`