carla-simulator/carla #6176: Carla lights give off two different errors (AttributeError and Aborted (core dumped))

angelomorgado, 2023-02-04:
CARLA version: 0.9.14
Platform/OS: Ubuntu 18

When I tried learning how lights work I ended up making a script to control them, however two errors appeared:

_terminate called after throwing an instance of 'std::runtime_error'
  what():  trying to operate on a destroyed actor; an actor's function was called, but the actor is already destroyed.
Aborted (core dumped)_

and

_my_lights = lmanager.get_light_group(carla.LightGroup.Street)
AttributeError: 'LightGroup' object has no attribute '__ iter __'_

So I tried using the code in the documentation and this is what happened:

\# Get the light manager and lights
lmanager = world.get_lightmanager()

\# Custom all lights - Gives off Aborted (core dumped)
mylights = lmanager.get_all_lights()
lmanager.turn_on(mylights)
lmanager.set_color(mylights,carla.Color(255,0,0))

\# Custom a group of lights - Gives off AttributeError
my_lights = lmanager.get_light_group(carla.LightGroup.Street)
lmanager.turn_on(my_lights)
lmanager.set_color(my_lights,carla.Color(255,0,0))

Is something wrong with my code? I tried looking for other problems similar to mine but I didn't find any

M-Colley, 2026-10-26:
Draft PR #10021 corrects the light-selector enum usage in the `ue58-dev` documentation; #10019 contains the `ue4-dev` counterpart. This addresses the documented selector/AttributeError subset. The destroyed-actor abort in this issue is not claimed fixed. Both drafts currently have merge conflicts, and live lighting validation remains unrun.

Blyron, 2026-11-12:
The documentation snippet that caused the `AttributeError` has been fixed, see the lights section of [core_world.md](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_world.md). `get_light_group()` takes a list of lights. Closing. If the abort on destroyed actors still happens with CARLA 1.0 (Unreal Engine 5.8, branch `ue58-dev`), please open a new issue with a minimal script.

closed 2026-11-12 by Blyron