carla-simulator/carla #6177: Expose vehicle RPM via ros-carla-msgs/msg/CarlaEgoVehicleStatus.msg

kis-balazs, 2023-02-04:
The [CarlaEgoVehicleInfo.msg](https://github.com/carla-simulator/ros-carla-msgs/blob/081fdcdc8a4f8d4ce936b792c4ef13aba3fd7478/msg/CarlaEgoVehicleInfo.msg) file exposes a **max_rpm** signal for each vehicle.

I believe it would be highly beneficial to provide, via the simulation, passed onto the ROS messages, the current **rpm** of the vehicle, via the [CarlaEgoVehicleStatus.msg](https://github.com/carla-simulator/ros-carla-msgs/blob/081fdcdc8a4f8d4ce936b792c4ef13aba3fd7478/msg/CarlaEgoVehicleStatus.msg), as a new signal.

Adding the value from the simulator would be part of the function of the ROS Bridge, [in function send_vehicle_msgs](https://github.com/carla-simulator/ros-bridge/blob/e9063d97ff5a724f76adbb1b852dc71da1dcfeec/carla_ros_bridge/src/carla_ros_bridge/ego_vehicle.py#L112).

Let's use the comment space to evaluate the complexity of the necessary modifications, and maybe consider adding this feature :)

kis-balazs, 2023-02-07:
Hey @MattRowe18!

How could we proceed with a discussion about this topic? The new signal would be useful to have.

glopezdiest, 2023-02-13:
While I agree that this would be a very interesting addition, I'd like to point out that it isn't ROS exclusive, as this isn't even exposed in CARLA's API, so we would be looking to modify the CARLA client

kis-balazs, 2023-02-13:
This is a very interesting angle @glopezdiest, I agree with the overall description of the problem.

I thought directly from my own use-cases' point of view, but yes, this is not something which is simply not exposed on ROS Bridge, rather not existing currently in CARLA API.
Might be good to rename and/or specify this in the issue.
What do you think?

glopezdiest, 2023-02-13:
I've added the feature request tag. I think we can keep the name as is

kis-balazs, 2023-02-13:
And how do you feel that we should continue?

Blyron, 2026-11-12:
Engine RPM is now exposed: `vehicle.get_telemetry_data().engine_rpm` (together with gear and other telemetry), see `carla.VehicleTelemetryData` in the [Python API reference](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/python_api.md). Closing as implemented.

closed 2026-11-12 by Blyron