carla-simulator/carla #6183: Running Carla without Graphics API

berndfrauenknecht, 2023-02-06:
I want to run Carla without rendering on a compute cluster. The cluster GPUs don't provide graphics APIs such as Vulkan or OpenGL. As I am not planning to render anyways, I don't care much that the GPUs cannot render. However, I couldn't get a Carla Server running in the first place as there is no Vulkan or OpenGL driver available. This is the case, even if I try to start Carla in no-rendering mode with: ./CarlaUE4.sh --no-rendering

Am I doing something wrong or is this a current limitation of Carla? In case it's a limitation, is it planned to have a setup that is able to run without any rendering support as is the case e.g. in MuJoCo in the future?
Thanks for your help!

CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04

berndfrauenknecht, 2023-02-13:
I am wondering whether it is possible to emulate a rendering interface with the CPU. However, I didn't find a working solution so far.

balckgxj, 2023-02-22:
I guess this is supported

Blyron, 2025-04-01:
You can run CarlaUE4.sh --nullrhi 
This should work.

closed 2025-04-01 by Blyron