carla-simulator/carla #6195: Carla 0.9.13 is Depth Camera "True Depth"?

EvanWangyifan, 2023-02-09:
CARLA version: 0.9.13
Platform/OS: Windows 11

I'm currently trying to get the "exact depth" from Carla simulations. But I found the depth camera's outputs are really accurate & similar no matter the weather/lighting condition. Since depth camera is normally really noisy in real world cases. Is this the way depth camera suppose to be in Carla? Is the depth camera mimics the exact distance? If not, how can I get the real "exact depth"? Thanks.

angryhen, 2023-02-13:
I have the same question. How can I get the real distance from depth image(output by depth camera)?

omarsayed7, 2024-12-23:
any help?

Blyron, 2026-11-12:
Yes. The depth camera returns the depth buffer of the scene (planar distance to the visible surface), with no noise model, so it is ground truth. See the depth camera section in [ref_sensors.md](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/ref_sensors.md). Add noise in post-processing if you need it.

closed 2026-11-12 by Blyron