carla-simulator/carla #6200: When co-simulation using carla and sumo(map imported from roadrunner), radar sensor cannot detect object.

lhd1996, 2023-02-11:
CARLA version: 0.9.13

Platform/OS: Windows

Problem you have experienced: When co-simulation using carla and sumo(map imported using roadrunner), radar sensor cannot detect object with velocity. Radar parameter has set to points_per_second:5000, still get total points about 300(with no velocity points detected). Does carla support radar sensor in customized map, or co-simulation?

What you expected to happen: Using radar sensor model in self-customized map with co-simulation of carla and sumo.

Steps to reproduce: import map from roadrunner, co-simulation, and collect data from radar sensor

Other information (documentation you consulted, workarounds you tried):

closed 2023-02-14 by lhd1996

dsz-hash, 2024-03-27:
Hi, have you solved the problem? I met the same problem, thank you very much for your reply.

minhaj6, 2025-04-13:
was the issue solved? @lhd1996