carla-simulator/carla #6201: Ue4 editor crash in carla 0.9.14 branch！

Lyyyyyyy, 2023-02-11:
if install the latest version of  ue4.26.2 then open the 'ue4editor.exe' in folder of binaries, you will get the editor crash alert：
e.g.
 "Assertion failed:ResourceTableFrameCounter == INDEX_NONE [File:C:/applications/UnrealEngine/Source/Runtime/Windows/D3D11RHI/Private/D3D11Texture.cpp][Line:2260]"

Lyyyyyyy, 2023-02-13:
I was rebuild ue4 after copy the additional Engine folder，but the ue4editor still crash。
The alert was that
 “ue4editor assertion failed:GPowerToRoughnessMaterialFunction 
[File:E/UnrealEngine/Engine/Source/Runtime/Engine/Private/Materials/Material.cpp][Line:522]
Cannot load PowerToRoughness”

glopezdiest, 2023-02-13:
Hey @Lyyyyyyy, not sure I'm understanding correctly but CARLA uses [it's own fork](https://github.com/CarlaUnreal/UnrealEngine) of Unreal Engine (based on 4.26). Are you using this version or are you using Unreal 4.26?

Lyyyyyyy, 2023-02-14:
Hi @glopezdiest , I'm use the lastest CARLA0.9.14 version, and use it's lastest fork(UnrealEngine4.26.2).  I guess the crash is relate that Window sdk version or GPU drive vesion， because it is ok on CARLA0.9.13.

Lyyyyyyy, 2023-02-20:
I had found the reason that ue4editor crash. I confused the additionnal map pack of Ue4NoEditor and building ue4editor from source code, So I put the files of engine folder of additionnal map at unreal4.26.2‘s folder.

closed 2023-02-20 by Lyyyyyyy

prabanjanKAN, 2024-02-11:
Anyone found the solution for this?

prabanjanKAN, 2024-02-13:
Launching the UE4Editor.exe file with -dx12 argument (like: UnrealEngine\Engine\Binaries\Win64> .\UE4Editor.exe -dx12) resolved  this error for me in windows 11. 

If -dx12 argument doesn't work, try -dx11 or -vulkan or -opengl.