carla-simulator/carla #6203: Carla multi-gpu PythonAPI example

ntscheppe, 2023-02-13:
Is there a python example for a multi gpu setup? 
Something like manual_control?

We are particularly interested in how to assign a sensor to a certain gpu and we couldnt find any information on this

Cheers

TurtleZhong, 2023-11-02:
Hi, Do you solve it?

Blyron, 2026-11-12:
Multi-GPU is transparent to the client: you connect to the primary server and spawn sensors as usual, so any example (for example `manual_control.py`) works unchanged. The setup is documented in [Multi-GPU](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_multigpu.md). Sensors are assigned to the secondary servers round-robin; choosing the GPU for a specific sensor is not supported. Please open a separate feature request if you need that.

closed 2026-11-12 by Blyron