carla-simulator/carla #6206: Error with make PythonAPI: make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1

Riyadh-kh, 2023-02-13:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.0.4

I'm facing an issue while trying to compile the PythonAPI client when I try to run:
make PythonAPI

the following is being displayed:
```
Using clang-10 as the CARLA compiler.
Setup.sh: llvm-8.0 already installed.
Setup.sh: Retrieving boost.
--2023-02-13 14:03:33--  https://boostorg.jfrog.io/artifactory/main/release/1.80.0/source/boost_1_80_0.tar.gz
Resolving boostorg.jfrog.io (boostorg.jfrog.io)... 34.216.251.87, 54.68.196.200, 54.71.165.42, ...
Connecting to boostorg.jfrog.io (boostorg.jfrog.io)|34.216.251.87|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com/aol-boostorg/filestore/4f/4ff012adaa7d8952913be05bf4233005de48289c?X-Artifactory-username=anonymous&X-Artifactory-repositoryKey=main&X-Artifactory-artifactPath=release%2F1.80.0%2Fsource%2Fboost_1_80_0.tar.gz&X-Artifactory-projectKey=default&x-jf-traceId=3313426a72981685&response-content-disposition=attachment%3Bfilename%3D%22boost_1_80_0.tar.gz%22&response-content-type=application%2Fx-gzip&X-Amz-Security-Token=FwoGZXIvYXdzEHMaDPDdc8OM%2Bi7Yik53VyKdBHOW27Y8LIZch%2F48Y6MzM2bKJ9dS0MluHd1iFzzw1gimi7yYBfXUoY9dufXRjvv7eLZ653oLWRUVl3%2FcWYmycG5wp8iokNvIV%2BF%2BANzuSwdmvpcaMbxDldJp%2Bqhrny7u0n%2BDb1nKld2seZ2AclFUc95RHpEeSwlqa3WkoEwNF%2BWojPooH1pfJD8L6DEGeU0V%2Foe2cGJ0Fmgn55mXzdGiNCLEvNM9NS4pJOKdqLEUT7W0otq7FqOktaGqMjPZGCEskPpCIXkDZ2Btk5yILed%2B9M4nJF2zJ65fOiBhwx6zWRI76CVvP5stR%2FT8BX6%2BD4qsoHNFlFtOn5Bel%2FjXi40Qm1r%2F8Opmal2SPAetjYEUybsJQJ3VFOO%2BjqTI38oS5gd9HENySQneKYsuI7quuQD%2F6ofibSvtoNR7BVk8Y1AXxWdvU%2BSi%2FlXWRUNd9VaAmg5z4q1cSwy1lLAneH6mYCoxHmVwql0%2BBJ4sBFNjyFJvzs4j6IMiDYYjsPNwn0aTmenzF5VCMK%2FspdntwEJQLEXc49e9K2Y5aK%2F%2F8033qSfVizcr3TSDtqHKPhH7LFF9CO7Xxq%2BY%2FlPBbfDmFu4kwgsO9wopDfYEW739ObGK1Qxpxo6XZoKCL%2B9FaMvoMyjBozGWiF868L33h%2Fc1%2BRne3Z3Iqi6mgeiwns6HeQ8KsvSCNrMCDNavMt39caRxO2CpXTpOh54Df5iqJNWl2QUzrnEozpKonwYyKh95HfUa7jFBhOTyIkn%2B36DobFk2peXcnxsz03JrgyrjVigLr2ijEo2pmg%3D%3D&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Date=20230213T100334Z&X-Amz-SignedHeaders=host&X-Amz-Expires=60&X-Amz-Credential=ASIASG3IHPL64QVEUUMI%2F20230213%2Fus-west-2%2Fs3%2Faws4_request&X-Amz-Signature=8dd9cc36a70a6c3013349b1d57a7d4eab3a3bd0d061d5ffb9dbd3079ff23e464 [following]
--2023-02-13 14:03:35--  https://jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com/aol-boostorg/filestore/4f/4ff012adaa7d8952913be05bf4233005de48289c?X-Artifactory-username=anonymous&X-Artifactory-repositoryKey=main&X-Artifactory-artifactPath=release%2F1.80.0%2Fsource%2Fboost_1_80_0.tar.gz&X-Artifactory-projectKey=default&x-jf-traceId=3313426a72981685&response-content-disposition=attachment%3Bfilename%3D%22boost_1_80_0.tar.gz%22&response-content-type=application%2Fx-gzip&X-Amz-Security-Token=FwoGZXIvYXdzEHMaDPDdc8OM%2Bi7Yik53VyKdBHOW27Y8LIZch%2F48Y6MzM2bKJ9dS0MluHd1iFzzw1gimi7yYBfXUoY9dufXRjvv7eLZ653oLWRUVl3%2FcWYmycG5wp8iokNvIV%2BF%2BANzuSwdmvpcaMbxDldJp%2Bqhrny7u0n%2BDb1nKld2seZ2AclFUc95RHpEeSwlqa3WkoEwNF%2BWojPooH1pfJD8L6DEGeU0V%2Foe2cGJ0Fmgn55mXzdGiNCLEvNM9NS4pJOKdqLEUT7W0otq7FqOktaGqMjPZGCEskPpCIXkDZ2Btk5yILed%2B9M4nJF2zJ65fOiBhwx6zWRI76CVvP5stR%2FT8BX6%2BD4qsoHNFlFtOn5Bel%2FjXi40Qm1r%2F8Opmal2SPAetjYEUybsJQJ3VFOO%2BjqTI38oS5gd9HENySQneKYsuI7quuQD%2F6ofibSvtoNR7BVk8Y1AXxWdvU%2BSi%2FlXWRUNd9VaAmg5z4q1cSwy1lLAneH6mYCoxHmVwql0%2BBJ4sBFNjyFJvzs4j6IMiDYYjsPNwn0aTmenzF5VCMK%2FspdntwEJQLEXc49e9K2Y5aK%2F%2F8033qSfVizcr3TSDtqHKPhH7LFF9CO7Xxq%2BY%2FlPBbfDmFu4kwgsO9wopDfYEW739ObGK1Qxpxo6XZoKCL%2B9FaMvoMyjBozGWiF868L33h%2Fc1%2BRne3Z3Iqi6mgeiwns6HeQ8KsvSCNrMCDNavMt39caRxO2CpXTpOh54Df5iqJNWl2QUzrnEozpKonwYyKh95HfUa7jFBhOTyIkn%2B36DobFk2peXcnxsz03JrgyrjVigLr2ijEo2pmg%3D%3D&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Date=20230213T100334Z&X-Amz-SignedHeaders=host&X-Amz-Expires=60&X-Amz-Credential=ASIASG3IHPL64QVEUUMI%2F20230213%2Fus-west-2%2Fs3%2Faws4_request&X-Amz-Signature=8dd9cc36a70a6c3013349b1d57a7d4eab3a3bd0d061d5ffb9dbd3079ff23e464
Resolving jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com (jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com)... 52.218.169.155, 52.92.243.225, 52.92.209.161, ...
Connecting to jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com (jfrog-prod-usw2-shared-oregon-main.s3.amazonaws.com)|52.218.169.155|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136670223 (130M) [application/x-gzip]
Saving to: ‘boost_1_80_0.tar.gz.13’

boost_1_80_0.tar.gz.13         100%[==================================================>] 130.34M  4.67MB/s    in 34s     

2023-02-13 14:04:10 (3.78 MB/s) - ‘boost_1_80_0.tar.gz.13’ saved [136670223/136670223]

Setup.sh: Extracting boost for Python 3.
Building B2 engine..

A C++11 capable compiler is required for building the B2 engine.
Toolset 'clang' does not appear to support C++11.

> clang++ -x c++ -std=c++11  check_cxx11.cpp
./tools/build/src/engine/build.sh: 120: clang++: not found

** Note, the C++11 capable compiler is _only_ required for building the B2
** engine. The B2 build system allows for using any C++ level and any other
** supported language and resource in your projects.

You can specify the toolset as the argument, i.e.:
    ./build.sh [options] gcc

Toolsets supported by this script are:
    acc, clang, como, gcc, intel-darwin, intel-linux, kcc, kylix, mipspro,
    pathscale, pgi, qcc, sun, sunpro, tru64cxx, vacpp

For any toolset you can override the path to the compiler with the '--cxx'
option. You can also use additional flags for the compiler with the
'--cxxflags' option.

A special toolset; cxx, is available which is used as a fallback when a more
specific toolset is not found and the cxx command is detected. The 'cxx'
toolset will use the '--cxx' and '--cxxflags' options, if present.

Options:
    --help                  Show this help message.
    --verbose               Show messages about what this script is doing.
    --debug                 Build b2 with debug information, and no
                            optimizations.
    --guess-toolset         Print the toolset we can detect for building.
    --cxx=CXX               The compiler exec to use instead of the detected
                            compiler exec.
    --cxxflags=CXXFLAGS     The compiler flags to use in addition to the
                            flags for the detected compiler.

Failed to build B2 build engine
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1
```

I would very much appreciate your help with this issue.

Yang2581, 2023-04-18:
I met this kind of problem but with a little bit difference. 

_clang-linux.link.dll bin.v2/libs/program_options/build/clang-linux-17.0/release/threading-multi/visibility-hidden/libboost_program_options.so.1.80.0
boost-install.generate-cmake-variant- bin.v2/libs/program_options/build/clang-linux-17.0/release/threading-multi/visibility-hidden/libboost_program_options-variant-shared.cmake
common.copy /home/dell/Carla0914/carla_0.9.14/Build/boost-1.80.0-c17-source/stage/lib/libboost_program_options.so.1.80.0
common.copy /home/dell/Carla0914/carla_0.9.14/Build/boost-1.80.0-c17-source/stage/lib/cmake/boost_program_options-1.80.0/libboost_program_options-variant-shared.cmake
ln-UNIX /home/dell/Carla0914/carla_0.9.14/Build/boost-1.80.0-c17-source/stage/lib/libboost_program_options.so
...failed updating 2 targets...
...skipped 15 targets...
...updated 261 targets...
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 1_

And the detail error log file is following below.
thank you very very very much for help me!
[error.log](https://github.com/carla-simulator/carla/files/11260048/error.log)

Yang2581, 2023-04-19:
> I met this kind of problem but with a little bit difference.
>
> _clang-linux.link.dll [...]

by the way, Platform/OS: Ubuntu 22.0.4

uni28, 2023-04-24:
Have you solved it yet? I had the same problem

uni28, 2023-04-24:
> Have you solved it yet? I had the same problem

zehranrgi, 2023-06-17:
If you face this problem, for me, I tried to get the previous version of Carla from the releases pages by downloading. The problem is because of the git. So, take the git folder from the latest version, paste it to your current version folder, and it should be worked. And, see if `git branch` giving anything to you.

Shiven110504, 2024-01-11:
I am also facing the same issue. Is anyone able to find a solution?

jimmyw404, 2024-02-23:
@Shiven110504  @Riyadh-kh What is the output of the command:

`clang++`

and 

`clang++ -v`

for you? For Ubuntu 20.04 It should be something like 

```
~$ clang++ -v
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-linux-gnu/7
...
```

StevenWICV, 2025-01-20:
> I met this kind of problem but with a little bit difference.
>
> _clang-linux.link.dll [...]

Have you solved this problem? Appreciate your reply