carla-simulator/carla #6207: How to get high quality images from RGB camera?

Yucong-Wang, 2023-02-13:
I get images from RGB camera like this. It is not very clear.
[image]

dalbom, 2024-08-29:
Is there any update on this issue? I'm having the same problem.

MarcelPiNacy-CVC, 2024-08-30:
Hi,

Sorry for the late reply, could you provide more information about your system and the version of CARLA you're using?

dalbom, 2024-08-30:
Hi, I'm using the [latest Carla](https://tiny.carla.org/carla-latest-windows) on Windows. I attached several cameras to an ego vehicle. The segmentation and depth cameras' outputs were okay, but the RGB camera's output is severely damaged. It seems like the models are of low polygons.

I saw from the other post that it seems like the LOD is somehow malfunctioning. But there was no solution on that post as well...