carla-simulator/carla #6213: Annotation of objects in Lidar range

AitorIglesias, 2023-02-13:
I am trying to annotate data from Carla Actors and Environment Objects (e.g. parked vehicles) detected by a Lidar Sensor. Is there any way to get the actors and Environment Objects in range of the Lidar? I am interested in annotating the bboxes and velocities, as well as the classes.

I have thought of different possibilities:

1. get all the actors in the world and check which ones are in the range of the Lidar. But this method seems to me very slow.

2. Place a Semantic Lidar in the same place of the normal Lidar, get the IDs of the Environment Objects from the raw_data of the Semantic Lidar. The problem is that the Environment Objects don't have the velocity field, is there any way to convert Environment Objects to Carla Actors? Or to get the ID of the Carla Actors from the Environment Objects?

I would also like to know if there is any other method.

DeepDuke, 2024-01-18:
Same problem. How did sovle it?

AitorIglesias, 2024-01-23:
Hi @DeepDuke, I found a solution but it is a bit slow.

First, I recommend you use the semantic LiDAR instead of the normal LiDAR. The normal LiDAR provides the intensity loss but you can calculate it easily whereas the semantic LiDAR provides the object tag and the id. If the id is equal to 0 then the point belongs to an environment object. You can filter the points with the tag of objects you don't want to annotate. Then filter the points with an id different from 0.

Now you have the points from the environment objects of interest in the point cloud. Now you need the candidates, for that get all the environment objects (only those with the tags of your interest) in the scene and knowing the LiDAR position and the range filter the bounding boxes of the objects.

Finally use the function contains of the [carla.BoundingBox](https://carla.readthedocs.io/en/latest/python_api/#carla.BoundingBox). I don't understand the purpose of the parameter transform but I previously transformed the points to the world coordinate system and then I used the function contains with an empty carla.Trasnform, it worked for me.

Additionally, you may know this but to annotate actors you can use the semantic LiDAR and it will provide the ids of the actors in the point cloud.

closed 2024-01-23 by AitorIglesias

DeepDuke, 2024-01-23:
@AitorIglesias Thank you very much. I finally got the label from the semantic lidar. But it seems we can not control the number of parked cars belonging to the environment.

AitorIglesias, 2024-01-23:
@DeepDuke Sorry but I've never tried to modify the number of parked cars, but I think you can disable some environment objects. That way you can reduce the number of parked cars. You can also spawn vehicles in parking slots to increase the number of parked cars.