carla-simulator/carla #6214: Using multi-GPU, can't get image info from callback function

dongkeyan, 2023-02-14:
1. CARLA version: 0.9.14
2. Platform/OS: 
-  primary server : windows
-  Secondary server : windows
3. Test Step:
-  primary server run command : CarlaUE4.exe -nullrhi carla-rpc-port=2000 -carla-primary-port=2002
- Secondary server run command : CarlaUE4.exe -carla-rpc-port=3000 -carla-primary-host=10.103.20.240 -carla-primary-port=2002 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=0
- primary server run python script to get image by sensor.listen
4. Problem you have experienced:
   I can't get image info from sensor.listen function .

balckgxj, 2023-02-24:
I had the same problem when primary server run cpp client to get image by sensor.listen
Any suggestions?

balckgxj, 2023-02-24:
I found that image info got from sensor.listen function succeed when the primary server and secondary server are on the same host. I don't know why

dongkeyan, 2023-02-24:
> I found that image info got from sensor.listen function succeed when the primary server and secondary server are on the same host. I don't know why

Hello，I guess that the firewall of windows is blocking the communication between primary server and secondary server.
You can change "localhost" to  remote PC ip in carla.Client function. And check the result.

I don't find the way to close windows firewall, if you are succeed,please share your method, thanks.

balckgxj, 2023-03-20:
Hello，I don't think the firewall caused the sensor callback exception because I disabled the firewall on both ubuntu machines while the exception was still not resolved。 
I don‘t know how to solve it.

veinlove55, 2024-03-27:
I think you guys forgot to call world.Tick(), I have faced the same phenomenon. In single GPU mode, no need to call world tick, however, in multi-gpu mode, we have to call it in client side.